opentx/opentx #8525: ACCESS receiver internal sensor ID

franz4949, 2021-05-23:
ACCESS allows to bind up to three receivers with telemetry, one of those receivers does actively transmit telemetry to the transmitter. A sensor search on the transmitter telemetry model page finds and shows the receiver internal sensors like RSSI, VFR, RxBT etc.. While the display of the Taranis X9d 2019 also shows an ID number behind each sensor, the smaller display of the X9LiteS does not show such ID number behind the sensor. By clicking on a sensor an Edit window opens where also the sensor ID then is shown, for the small display this is the only option to see the sensor ID. For the bigger display of the Taranis X9d 2019 you will however find, that the sensor ID in the Edit window is different than the sensor ID shown directly behind each sensor on the telemetry model page. Both values for the same sensor differ while they should be the same. It seems that the sensor IDs in the sensor Edit window are wrong.

Tested with Taranis X9d 2019, OTx 2.3.10, RX6R, latest LBT ACCESS firmware and Taranis X9LiteS, OTx 2.3.11, RX6R, latest FCC ACCESS firmware.

Hope that my description of finding is understandable so far.

franz4949, 2021-05-23:
Here you can see the RX6R bound as Rx2 :
The two photos show the X9d display and you can read e.g. RSSI sensor ID 56. If you open the RSSI Edit window it shows the sensor ID as F101  25. Thus two different sensor IDs are shown for same sensor. Unfortunately I don't have a photo of the X9d 2019  RSSI sensor Edit window now. If you need or want I can arrange for getting such photo.

[image]

[image]

mpjf01, 2021-05-24:
See #8515 and others

franz4949, 2021-05-24:
@mpjf01 Thank you for your link, I have read your #8515 issue.

I just wonder why OTx shows for same receiver internal sensor (here as example RSSI) different ID numbers, ID=56 in TELEMETRY window and when RSSI Edit window is opened it shows there ID=25.
Either OTx reads the sensor ID from different sources (and one source provides wrong values), or maybe OTx just takes standard sensor ID values in the Edit window without any reading the actual ID number.
It is fact that the sensor IDs are linked and vary with the position of receiver binding (Rx1, Rx2, Rx3). This can easily been checked by just binding same receiver in either position, the big X9d display shows the change of sensor ID. This works with just a single receiver bound on respective Rxx-place, no SPort, no SBus connection needed for this simple test.

Would appreciate some clarification here from an OTx dev.

MarkusPfaff, 2021-05-27:
Go into detailed screen for the sensor and try to change the physical ID. You can have values from 1 to 255 there. That is because there are 8 blocks of 32 sensor values each. It seems that each receiver has one block and the external module has 4 and the old module seems to have one (even if it is not installed).

franz4949, 2021-05-28:
I do not understand why I should do this. As soon as I touch the sensor ID in the Edit window and change its number, the transmission of sensor data stops. Shall I now go through all numbers from 1 to 256?

I just want to understand why OTx shows different IDs for same sensor dependent into which display window I look (Telemetry page or sensor Edit page). To my understanding only one value can be the correct one.

MarkusPfaff, 2021-05-28:
physical IDs can have the values 1 to 28 only (as far as I know). OpenTx uses the value set 0 to 31 (or 1 to 32) for the complete set. ACCESS allows for up to three receivers to be bound to a Tx. This gives you 3 times the range from 0 to 31 for every source.  If you go through the IDs in the detailed telemetry screen you will note that you have in sum 8 times that range and the "source" field will change when the 32 value limits are crossed. OpenTx seems to make a single number in the overview screen from the physical ID in combination with the "source"

Of course the displayed ID in the overview screen is inconsistent with the value in the detailed screen anyway in that it shows the value-1. But this is an old bug still existing. 

More details can be found in this [thread in the german FrSky forum](https://frsky-forum.de/index.php?thread/5410-x12s-upgrade-auf-access-mit-isrm-nachr%C3%BCstmodul-f%C3%BCr-die-x10-x10s-sensoren-id-ab-1/).