opentx/opentx #8526: Backlight turns off when changing model and RX is still powered

nmaggioni, 2021-05-25:
**Describe the bug**
When changing model while a receiver is still connected, the backlight is turned off and the onscreen alert can't be seen.

**To Reproduce**
1. Select a model and ensure that a receiver is connected.
2. Without powering down the RX, enter the model selection menu and try to switch to a different one.
3. The backlight gets turned off and the "Model warning - model still powered" alert is displayed.
4. Pressing any button other then Enter or Exit has no effect and the backlight stays off.
5. Pressing either Enter or Exit dismisses the dialog with the proper action and the backlight is turned back on.

**Expected behavior**
The backlight should stay on while displaying the warning.

**Screenshots**
N/A

**General information**
 - OpenTX version: 2.3.11-otx (4328055e)
 - Hardware: FrSky Horus X10S
 - OpenTX Companion: N/A
 - TX modules: ImmersionRC Ghost

3djc, 2021-05-25:
Clearly not happening here, please post your otx

nmaggioni, 2021-05-25:
Here's a slimmed down config with the two models I'm currently using: [8526_models.zip](https://github.com/opentx/opentx/files/6538478/8526_models.zip)

I can reproduce it consistently, let me know what else I can do to help.

3djc, 2021-05-25:
Your ON brighness is all the way down, I think thats why

nmaggioni, 2021-05-25:
Further testing indeed hints that the warning dialog uses the OFF brightness value, not the ON. I've updated the steps to reproduce.