opentx/opentx #8528: Wiki Recommends Docker Despite deprecation of dockerfile

Marsfan, 2021-05-27:
The wiki says that Docker is the recommended way to build OpenTX for reliable builds (here: https://github.com/opentx/opentx/wiki/Compiling-OpenTx). 

However, in #7308 @3djc says that the dockerfile was deliberately removed as it was intended to no longer be used. So which is it? I need to build the project my self as I need version 2.3.9 and the companion does not allow building previous versions.

alm-nl, 2021-06-05:
You could also use https://github.com/vitas/opentx-docker-build or follow the procedure written here: https://www.rcgroups.com/forums/showthread.php?3824801-Blog-1-Step-by-step-guide-to-building-(custom)-OpenTX-firmware-under-Ubuntu-Linux
Both work for me.

nmaggioni, 2021-06-17:
#8001 works fine but it has been closed with no explanations. I'm sure there are valid reasons behind this choice, but I wish contributors' work wasn't just thrown away like that.

alm-nl, 2021-06-17:
I can imagine the devs only want software directly related to OpenTX in this repo because a docker image they don't use adds up to the total size without the devs using it? Why don't you create your own repo then (if you didn't already that is)?

nmaggioni, 2021-06-17:
@alm-nl How is a spec needed to build a software not directly related to it? A Dockerfile was included in the repo until not long ago, and I think you haven't even taken a peek at the related PR since size isn't exactly an issue when we're talking about a dozen lines of text. The same goes for "creating my own repo": where do you think my PRs comes from? :)

raphaelcoeffic, 2021-06-17:
@nmaggioni this is more a question of philosophy, I guess. You might be able to use this as well: https://github.com/EdgeTX/build-edgetx/tree/2.3

alm-nl, 2021-06-17:
I think the answer is here: https://github.com/opentx/opentx/issues/7308#issuecomment-671174218

Marsfan, 2021-06-19:
My problem is that the wiki still recommends using the docker file, despite the docker file no longer existing.

pfeerick, 2021-07-06:
> It is no accident. This file is not required to build OpenTX, we have no use for it, and do not whish to maintain it

I can agree with the sentiment that it's not needed by the devs, and thus they don't want to maintain it. Plus, it's there in the repo history of anyone who wants to retrieve it. But I'm surprised they don't use docker themselves or want to encourage people to use it, in order to reduce the number of "why doesn't this build on my system" issues that result from a non-automated / virtualised build environment. This is the exact same reason other projects are moving towards PlatformIO for building stuff... it automates the build environment setup, and if there is a problem on the PlatformIO side of things, you to them for help... meaning it's _not_ the devs problem to deal with or avoid. The same would be the case here. Especially since it is a single solution that works cross-platform ;)