opentx/opentx #8531: LUA set to "ON" but checked for "YES" in CMakeLists.txt

pagrey, 2021-05-28:
I noticed "custom scripts" are missing on the T-Lite build and lua is not in the firmware build options although LUA is set to ON. It appears although LUA is set to ON on line 40 of the default CMakeLists.txt it is later checked for YES on line 174 so LUA_MODEL_SCRIPTS isn't defined unless you change LUA to YES before compiling. I don't know if that's how it is supposed to be or not. It also looks like FAI on line 215 uses YES instead of ON as well.

3djc, 2021-05-28:
2.3.12 nightly downloaded with companion:
[image]

closed 2021-05-28 by 3djc

pagrey, 2021-05-28:
You don't publish how you make your nighties so it doesn't have allot to do with the code here. Your CMakeCache has been changed by your nightly build process in a way that is not available for anybody to see as far as I know.

If you type "cmake -D PCB=X7 -D PCBREV=TLITE ../" you will get LUA="ON" and lua will not be enabled. That is what I am commenting on. I know you can set LUA="YES" and enable it. I did not say lua is not available, I said ON means OFF in the default config.