opentx/opentx #8539: Convenient GPS format for lost drone location

MrMap, 2021-05-31:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
Losing a drone in the terrain can be a bit stressful. Thanks to the GPS telemetry feature and good beepers the odds are fair to locate and recover it by reading the last reported coordinates from the radio telemetry display and enter it to google maps on the mobile phone. However, I found two hurdles that made this unexpectedly difficult during the last time I found myself in such a situation:

1. The coordinates were very difficult to read from the radio display (RM TX16S). I had lost telemetry contact and so the display digits grayed out, making readout very very challenging - right when I really needed clear and easily readable digits.
2. The coordinates were presented in DMS format, presenting the challenge of entering the special characters and format 100% correct into the google maps search field (it is not forgiving at all and I did not succeed).

**Describe the solution you'd like**
Purpose being making the lost drone process as easy and streamlined as possible, using the radio TX and google maps:

1. Add simplified coordinate formats as options besides DMS and NMEA. By simplified I mean formats not using special characters like degree, minute, second symbols - e.g: 53.339688, -6.236688        Maybe even the google plus code format?
In essence: The most practical format to read and enter coordinates into google maps.

2. And maybe consider offer an user option to make latest reported GPS coordinates extra bright and clear on the display upon lost telemetry i.s.o. the opposite - gray out.

**Describe alternatives you've considered**
Since I had the in field experience I have made two setting changes to simplify the lost drone finding process:

1. I have checked "shadow" on the relevant telemetry presentation data fields. Makes it much easier to read even if it would be grayed out due to no incoming telemetry data.
2. In the radio system settings I have changed coordinates format from DMS to NMEA. Considerably easier to read and enter correctly into google maps search field. (But could be made even easier...)

**Additional context**
Add any other context or screenshots about the feature request here.

lshems, 2021-05-31:
Sounds like a widget. You can even have the last position expressed in meters distance with heading from internal GPS or start position.

MrMap, 2021-05-31:
> Sounds like a widget. You can even have the last position expressed in meters distance with heading from internal GPS or start position.

Sorry, I don´t follow. How would this simplify locating the lost drone using google map?

lshems, 2021-05-31:
Well,it gives you a direct hint on where it is related to your current position.

That can help check if you entered the coordinates that you can show in such a widget in big bold fat letters correctly and you are not heading the wrong way.

Nevermind. Just showing anything from a sensor in any format is normally resolved by a widget. That's what they are for.

MrMap, 2021-06-01:
Ok, seems like a widget - new one for this purpose or an update of the one already in use - is the way to go.
Anyway, I think the feature itself would be recognized as useful by the community - sooner or later we all may experience an "unscheduled distant landing".

I don´t have the skills to contribute with code but I volunteer to assist with ideas and testing.

lshems, 2021-06-01:
Make a mock-up and description and post it here.

I'll do a Liz widget for this one.

lshems, 2021-06-01:
Liz is LUA

3djc, 2021-06-01:
Idealy, a qr code would be perfect, I think LUA libs do exist

lshems, 2021-06-01:
What is a qr code, and what LUA libs do we need/can we use?

MrMap, 2021-06-01:
**QR-code - of course! Brilliant idea!**
https://en.wikipedia.org/wiki/QR_code

And maybe below the QR, a text string displaying the telemetry time stamp of the position displayed?
*Maybe* also - the latest received (if telemetry lost) GPS speed, course, altitude, number of satellites, vertical speed and G-force?

lshems, 2021-06-01:
Yes!!  Very nice suggestion, and the perfect way to interact!

lshems, 2021-06-01:
I'll look into it.

estebanavv, 2021-09-07:
You might want to consider using Google Plus codes ([Open Location Code] s this was implemented in Ardupilot too. (https://en.wikipedia.org/wiki/Open_Location_Code)) [Miami Mike](https://www.rcgroups.com/forums/showthread.php?2823315-OpenTx-2-x-x/page408) already created some code.
[GooglePlusCode.zip](https://github.com/opentx/opentx/files/7121297/GooglePlusCode.zip)

ChrisOhara57, 2021-10-31:
I took the liberty to alter the code of Miami Mike.
In this version it permanently stores the last received GPS coordinates by background function.
So it can display the last position as Google+Code even after connection to receiver is lost.
Works as telemetry script on Taranis.
[gpluss.zip](https://github.com/opentx/opentx/files/7449821/gpluss.zip)

rotorman, 2022-04-30:
Moschotto's LUA GPS telemetry widget at least partially solves the original issue: https://github.com/moschotto/OpenTX_GPS_Telemetry