opentx/opentx #8540: Emergency mod setting up widgets

victorcapanema, 2021-06-03:
**Describe the bug**
Every time I scroll through the list of widgets that is a quarter the size of the screen the radio enters emergency mode.

**To Reproduce**
Steps in the video.

**Expected behavior**
Not trigger emergency mode.

**Screenshots**
https://user-images.githubusercontent.com/47992324/120645456-d382f400-c44e-11eb-8ac4-52d6b5dcbe77.mp4.

**General information**
 - OpenTX version 2.3.11
 - Hardware TX16S

fabo137, 2021-06-03:
Checked on my TX16S - You're right, have the same issue.

I have the issue when i deactivate for instance the trims or any other side element on the telemetry screen.

victorcapanema, 2021-06-03:
> I cannot reproduce with a 2.3.11 otx and our sd content.
> 
> Delete all content /WIDGETS and put only those in https://downloads.open-tx.org/2.3/release/sdcard/opentx-tx16s/sdcard-480x272-2.3V0035.zip

Done it. The first launch got emergency mode, after launch again it stopped. I had one gps widget that overlaps the size of that frame, it looks like oversize things cause this.

BlueBlue58, 2021-06-03:
> **Describe the bug**
> Every time I scroll through the list of widgets that is a quarter the size of the screen the radio enters emergency mode.
>
> **To Reproduce**
> Steps in the video.
>
> **Expected [...]

I have had this same issue in the last 3 days but mine occurred three times when I was trying to bind a receiver on my Horus X10 Exoress. Namely when scrolling through the list of makes of receivers and when it reaches the AFHDS2A it crashes and goes to the Emergency Mode. If I switch off and back on then the radio seems to be fine. It was only triggered when I tried the AFHDS2A channel if that makes sense?