opentx/opentx #8541: Windows Companion can't be used without corrupting the Trainer Mode: field

jims1234, 2021-06-04:
You will see that I have reported this problem months ago as a part of issue #7873 which was I think mis-interpreted and re-written by others in authority here.  
I frankly do NOT care about the Trainer Mode Simulation issue with the Trainer in TRAINER ON status. That's never worked. That problem is still there in Companion too, but is I feel very low pri.. 

This issue is serious since it means Companion users using this wireless training mode can not edit ANY model in companion and Write it back to the Radio without forcing him/her to need to check and edit the Trainer Mode parameter and change it back to: "MULTI/Master" any time you edit ANY other model for any reason.     

The Windows Companion is a wonderful tool for editing and backing up models however for those of us that like to do Wireless Training using the Trainer Mode type : MULTI/MASTER  find this mode field is corrupted whenever models programmed to support it are read in from the Radio, and also in the other direction, when a user Writes Models and Settings and has pre-set that Trainer Mode:  to "Multi/Master" the Model Setup Page inside the Radio will show that this field was corrupted during the write and that the Model Setup Page displays the following Trainer Mode: "eck RSSI on Shutdo"  instead.       
**To Reproduce**
Steps to reproduce the behavior:
1. In Companion, Go to any Model with the Wireless Training mode set to "Multi/Master" and then Write Models and Settings to the radio.  Select Models then scroll up to the Trainer Mode field.  You will see "eck RSSI on shutdo" instead and must change it back to Master /Multi to re-enable the basic Wireless Trainer function to work.
and also...
2. In Companion connect any radio with Wireless Training mode Master Multi enabled. Select Read Models and Settings then go to that model's Radio Setup Page and look at the new value in this field..  It will read Master/Bluetooth instead of what was last set inside the Radio,  and must be edited back to "Multi/Master" to fixit.

These behaviors 1. and 2. above do NOT produce these errors as stated 100% of the time only approximately 98% of the time..why it seems to work some times is not clear.  

**Expected behavior**
One would expect that when he/she makes a change on either the Radio side or the Companion side and re-Writes, or re-Reads  it back across the USB interface that the parameter should not change to some other value as it is doing now.    

If applicable, add screenshots to help explain your problem... Ok I'll try to get a couple 
 - OpenTX version: 2.3.11
 - Hardware: Radiomaster TX16S
 - OpenTX Companion:  2.3.11 on Windows10 64bit 20H01)

3djc, 2021-06-05:
I'm not seing that. If I setup a model with trainer/multi, read it into companion, edit it (NOT changing trainer mode), write it back, it is still trainer/multi.

There is an issue in what is displayed in companion as trainer mode, not sure it can be easily fixed

jims1234, 2021-06-06:
@3djc   Thanks for taking a look at this issue. I don't know if anyone else can duplicate these, but both happen to me almost every time I use Companion, (so as a work-around every time I call up a model in the Radio before any Wireless Training flight I have to check for the "eck RSSI on shutdo" string and set it manually back to "Multi/Master: or the Training link won't work.  

I believe you are saying you do see issue 2. but it will be hard to fix.. correct?   

I believe 1. and 2. started for me when Trainer Multi was 1st released.  Could it be that having approximately 20 models is too many?  If there's anything I can do to try and answer questions about the issue or help ID it.. Please do ask.  Meanwhile I will ask on RCGroups if I'm alone or if others using are seeing these Companion issues with Trainer Mode : Multi/Master too.    Thanks again..

3djc, 2021-06-06:
You are not reading what I wrote…

For now, do NOT edit the trainer field in companion, it will not get changed if you read/write if you don’t touch it

jims1234, 2021-06-07:
Ok great, .. I'll wait for 2.3.12

jims1234, 2021-06-07:
3djc,

I read it alright but I’m apparently not very good at reading between the lines or  understanding what’s going on under the hood..

 

I do understand that you plan to fixit in 2.3.12 .. that’s great ..

 

I do understand that I cannot use Companion at all on ANY model in the interim.. because.. if I use it to mod ANY model it neg impacts all with Trainer Multi capability randomly. Its neg-impact is a lot broader than I believe you understand. Fortunately the work around is simple and now a knee-jerk reaction to manually edit it in the Radio and fixit. 

 

I do NOT understand how, and why this issue was so difficult to get any visibility and to address since it’s been many many months that I’ve repeatedly tried to flag it.. 

 

Why? Because  I love flying with my Grandkids and its worth the effort.. I also believe you when you say you aren’t seeing the issue occur at all.. I suspect it’s because my multi-model environment is different, and I’m using an EXTERNAL MPM to host all my models rather than the INT RF MPM  that may be the norm.  I don’t know why it doesn’t corrupt the field EVERY time either. .    

 

I hope you understand how much I DO VER MUCH appreciate your efforts and all the dev’s back there and I try to support you in any way I can.

 

Best Wishes,,

jimS

PR #8545 (2021-06-07): Fixes trainer setup in Companion
Fixes #8541
Exposes missing ppm fields for Slave/Jack to match radio
Adds item model and toString for trainer mode
Remove static trainer modes from ui definition and use filtered item model
Print trainer mode use toString function.
Note: only tested using Companion Simulator for X9D+, TX16S, X12S, X7 and a couple of others with and without bluetooth trainer
closed 2021-06-07 by bsongis