opentx/opentx #8542: Add warning in Companion prior to write to SD card

mpjf01, 2021-06-05:
Many users have reported accidentally deleting all models on their transmitter. The issue usually arises when they are trying to add one more model supplied on an .otx file from another user. They either just write that one model to their radio thinking that it will be added to what's there already or have the wrong window active when doing the write command.

It would be helpful if whenever the "write models and settings to radio" command was enacted, for Companion to give a popup warning stating that this command will overwrite all existing models on the radio with whatever is showing in the active screen, do you want to proceed yes - no.

3djc, 2021-06-05:
I'm not opposed to the idea, especialy if we have a tickbox to not show this warning again, but honestly, not sure it  will improve much

elecpower, 2021-06-05:
@3djc agree. I do not know how many times I have come across users in many other apps clicking without reading such prompts and then complaining that there should have been Are you really really sure? prompt. That being said a prompt until you tell me to stop is still not a bad idea. There are a few places in Companion where the prompts can get annoying after a while or no prompt at all because it would be annoying.

mpjf01, 2021-06-07:
Most of the users that I have come across who have deleted all of their models are inexperienced with computers and/or Companion and they do tend to read warning popups when they are presented. They usually don't write in forums or feed back their problems except personally.

The users who click OK without reading popups are usually the more experienced and would mostly be unlikely to make the error I am trying to prevent. 

It would seem to me that the better option is to have a popup warning but, as suggested, add the option not to see it again. That way everyone is catered for.

elecpower, 2021-06-10:
Is this suitable?

[image]

3djc, 2021-06-10:
Looks perfect to me

mpjf01, 2021-06-11:
Perfect, thanks.

PR #8550 (2021-09-07): Confirmation prompt before writing models and settings to radio
Fixes #8542
The prompt can be turned off
closed 2021-09-07 by 3djc