opentx/opentx #8546: Menue navigation cursor out of screen on T-Lite

0815Creeper, 2021-06-07:
**Describe the bug**
on the jumper T-lite (maybe also on other radios) the cursor can get "lost" outside the page.

**To Reproduce**
Simulate Radio in companion.
go to e.g. spectial functions or e.g. model setup, switch warning.
Scroll to the right or left multiple times, scrolling back requires more than one click. 
There is no "wall" but instead you can scroll further without hitting the wall/scroll through to the other side as you would on radios like x9e or x9d

**General information**
 - OpenTX version N510
 - Hardware Jumper T-Lite

0815Creeper, 2021-06-08:
I tried to show it in these videos:
https://youtube.com/shorts/anidbfumckY?feature=share
https://youtube.com/shorts/le58XiDSXDw?feature=share

0815Creeper, 2021-06-10:
it does not seem to be a bug like cornercase, but more like a general problem in the way the menue items are navigated

0815Creeper, 2021-06-13:
> You will have to be more specific on where those things occur

Still not specific enough?

3djc, 2021-06-13:
No, we ahev reproduced it, it will be adressed when time allows it

0815Creeper, 2022-04-16:
Well, I hope there will be time when one of the next releases is beeing published ;)