opentx/opentx #8551: Open TX 2.3.11 and Hours X12S

505ray, 2021-06-11:
updated to OTX 2.3.11 
loaded on computer fine,
try to update Hours X12S .. it seems to try to do so
but I get an error message of ;
Cannot write to L;//FIRMWARE.BIN (reason Unknown error)
TX is currently at 2.3.10-otx
Don't know why... ??
thoughts ?

3djc, 2021-06-11:
Sorry but this place is only for tracking otx bugs, not usage issues. Please go to community place like rcgroups or openrcforum for community support

closed 2021-06-11 by 3djc

505ray, 2021-06-11:
Sorry,  can you please suggest a site to use  ??

pfeerick, 2021-06-12:
You were given two suggestions above... rc groups and openrcforum. However, you most likely tried to copy the file to the wrong drive - you should be copying it onto the SD card drive, not the internal memory.