opentx/opentx #8554: Crossfire Tx not longer connect receiver after updating to 2.3.12

sylphio, 2021-06-17:
**Describe the bug**
After updating from opentx 2.3.9 to otx 2.3.12, my TBS CROSSFIRE TX module is no longer able to connect to crossfire nano receiver (led on the receiver stay red,  normally once a connection is establish the led become green).

Just after the TX is powered, during 1sec the RX led become green but afterward the connection is lost again.
On the TX module lcd display, during the first second after powering the TX,  CRSF V2 is noted as the RC input but as soon it automatically switch to CRSFShot the connection is lost.

I tested with two different receiver, one nano on a INAV build and one nano rx on betaflight both have the same issues.

To confirm it is OTX related, I flashed back otx 2.3.9 on my transmitter and the issue disappear.

**To Reproduce**
Just power previously binded a Crossfire TX and a receiver after updating otx to 2.3.12

**Expected behavior**
The led on the receiver should become green and a valid connection should be established.

**General information**
 OpenTX version 2.3.12 
 Crossfire version 4.11 TX & RX (latest)
 Radiomaster TX16S
 TBS CROSSFIRE TX (the big 2W one)
 TBS receiver nano

3djc, 2021-06-17:
It works fine here, I guess your issue must be related to crsf rxnum introduced in 2.3.10

closed 2021-06-17 by 3djc

sylphio, 2021-06-17:
@3djc I was wondering if my issue was related to the tbs cloud/multibind who is enabled on both of my transmitter? It is not used by a lot of people.

Do you have a suggestion on how I can fix that?
Thank you

sylphio, 2021-06-17:
Ha effectively, it can be the rxnum, I use non-zero number for both of them.