opentx/opentx #8557: Taranis multifunction switch is backwards in companion simulator vs. on Radio

patl1, 2021-06-17:
I am using a Taranis X9B+ with latest firmware and companion 2.3.12. I have a 6 position multifunction switch installed as S3 and it works as expected in flight for my drone but when I run the simulator in Companion, the switch positions are reversed from that the radio does

3djc, 2021-06-17:
Companion switch positions are logical, not physical and does not represent how they are on the radio

closed 2021-06-17 by 3djc

patl1, 2021-06-17:
OK thanks. So how do I go about correcting the reversed 6 positions of the multi function switch? The other switch positions on the radio are correct for some reason.
thank you

patl1, 2021-06-22:
Interesting that the other logical switches on Companion move in the same direction as the physical ones on the radio (i.e. rotation directions of S1 and S2 match the directions of the switches on the radio). Seems that the programming in companion should represent the rotational detent positions of  the actual6 position  switch. Why is this thinking incorrect. Did I just get lucky on S1 and S2 that they rotate the same direction?