opentx/opentx #8559: Feature: possibilty to comment values ('any' value that can be calibrated by the user), mixer lines

FranzDurst, 2021-06-18:
Hi folks,
It would be really helpfull if it would be possible to comment the functions (mixer lines, logical switches, etc.) and calibration values  (GV's, and any other number for tuning the model). Like you and I would do when coding something.
Many models ..., some special solutions etc., maybe some remember everything why what was done, These are probably minority.
This would be a feature, mainly used in companion I think. Don't know if there is a way to make this possible and usefull on Tx. too.
Maybe comment pops up on mouse hoover like it is the case for logical switches, and something empty already pops up on curves too.
Thank you for this first class system!
FranzD

bigtwin, 2021-06-18:
Sounds familiar to https://github.com/opentx/opentx/issues/7645
Unfortunately not much of a chance ... :-(

elecpower, 2021-06-18:
Yaml format may address this requirement but likely only for radios that store models on sd card.
Eeprom limit for radios like x9d+ has prevented this feature and the need to be able to transfer models between radios.

FranzDurst, 2021-06-19:
@bigtwin : yes, thank you for this note: duplicates #7645 .
So, seems not very probable to happen ...