opentx/opentx #8560: Support for M-Link Telemetry

Hartinger, 2021-06-19:
OoenTX Version 2.3.12 has introduced M-Link telemetry support for the Multi-Module. I fly with genuine Multiplex M-Link Modules and it would be very nice if OpenTX could support M-Link telemetry data from those modules as well. I and others are currently using an Arduino with a converter that I have written in C, which transforms the telemetry data from the module into the D8 protocol. But obviously, it would be a lot more elegant if OpenTX could directly process the M-Link telemetry data.

3djc, 2021-06-19:
Woulnd't it be simpler to use a multi module ?

Hartinger, 2021-06-19:
It would be simpler but illegal in Germany.

wimalopaan, 2021-06-19:
According to which EU regulation do you think it is illegal?

Hartinger, 2021-06-19:
It has no value to discuss this here in ny detail, as I am not a legal expert.
Just that much:
If you want to operate a transmitting device that you have imported into the EU you need a declaration of conformity. There is no official importer of multi-modules (as far as I am aware) that you could buy from, and who would provide this declaration. So I myself would be the importer and would have to generate this document. I will certainly not do that, as I have no means to verify that the device fullfills all applicable regulations. (This is normally certified by an authorised test body and costs quite some money.)

Hartinger, 2021-06-19:
Sorry, I closed this issue inadvertantly and have reopened it now.