opentx/opentx #8561: Companion 2.3.12 and MAC OSX Mojave

bigtwin, 2021-06-19:
I reporting this as a bug as I could not believe that it is supposed as I found it ...

I just wanted to update my Companion from 2.3.11 to 2.3.12 and found it not supported with MAC OSX 10.14.6 (Mojave). It is not even 2 years ago, that Mojave was substituted by Catalina and I cannot believe that OpenTX is that progressive ...

Please provide a build which is does not require a MAC OSX later than 2 Years ...

If this is not an option, would it be possible to maintain OTX 2.3.12 with Companion 2.3.11?

kilrah, 2021-06-21:
Use a nightly, even the current one works. It's unknown why the release built the same way doesn't, but we're looking into it.

MacFlieger, 2021-06-21:
I have the same problem with macOS High Sierra.
I have looked at this nightlies:
https://downloads.open-tx.org/2.3/nightlies/companion/macosx/
The last nightly of 2.3.12 is from 14.5.2021 and is working with older macOS versions
The newer macOS nightlies since 16.6.2021 of 2.3.13 are completely broken (637bytes long)

3djc, 2021-06-21:
Get mac nightlies from here : https://github.com/opentx/opentx/releases/tag/nightly

bigtwin, 2021-06-24:
While the nightly build **2.3.12N520 worked fine**, the just released 2.3.13 official release from Companion again does not run on MAC OSX Mojave (and probably High Sierra).

:-(

timtuxworth, 2021-06-24:
Perhaps you should update your Mac @bigtwin - I'm running Big Sur (11.4) and it works just fine.

bigtwin, 2021-06-24:
Please read my initial Post and the thread before writing something like this. This thread is already beyond of your breaking new idea. ;-) 
It is confirmed not the intention of OTX to just and only run on the very latest OSX. 

And again the nightly builds work all. Just the official release do not. It seems like there is a little difference in the way the Nightlies and the Officials are compiled.

MacFlieger, 2021-06-25:
@timtuxworth:
High Sierra is the highest macOS which is running on my Mac. I cannot update to BigSur.
I would understand the problem, if Companion uses a feature of the newer OS. But as the nightlies are working with the old macOS, it seems to be a problem of the building process.

3djc, 2021-06-25:
It probably is, but since none of the devs have mac expertise, or even a own mac, it will require somebody with the right skill and equipement willing to work on this. This is typicaly the spirit of opensource...

bigtwin, 2021-06-25:
So what about the one, who did the official Mac Build for 2.3.11 And who is currently building the Nightlies? Obviously everything tis OK there.

Let me know if I can be of help here. I do have IT Skill. I even have some Dev-Background but this is from many years ago. So if you guys want me to action something on my MAC, I would probably need good instructions ...

I am in the same position as @MacFlieger. If I would need to update my MAC it causes a lot of efforts and (license-) costs. And as said - _High Sierra_ and _Mojave_  ought to be young enough for OTX :-)

kilrah, 2021-06-25:
Nightlies are built the same as the release, that's the point. It's been set up in github actions, you can look at it and see if you spot a significant difference, we didn't. 
https://github.com/opentx/opentx/actions

The machine that was bulding older versions that was provided by a user/contributor has gone down and we could not get contact again, hence the change in build process.

bigtwin, 2021-06-25:
I would be happy to help out with some action on my MAC. But I cannot actively work with GitHub nor do I have sufficient development skill to dive into it. If I can help, I would need instructions according "Building Companion for Dummies" ;-)

MacFlieger, 2021-06-25:
There must be a difference in the two build scripts, cause the nightlies all works and the release doesn't.
I have tried a little bit and found something interesting.

First of all a comment for the people, who are not using a mac and can't test it directly: The release-version doesn't crash or don't work, but the finder shows a forbidden sign over the normal icon and when I try to start the app, the finder gives an error, that a newer macOS version ist needed. The finder is not trying to start the companion because of some misleading/wrong information.

Now for the interesting things:
1. I tried to start the app from a terminal with the command "open". Now the app is really tried to start and an error occurs: "LSOpenURLsWithRole() failed with error -10825 for the file /Applications/OpenTX Companion 2.3.13.app."
2. I tried to start the executable (the file "companion" in the subfolder "Contents/MacOS") directly in the terminal and success: Companion launches. But if I try to load an ".otx"-file, it crashes with "dyld: Symbol not found: ____chkstk_darwin" and the comment, that the executable "companion" ist built for Mac OS X 10.15.

I have compared the building scripts and also have found no difference. Only two more options "-DALLOW_NIGHTLY_BUILDS=YES -DTEST_BUILD_WARNING=YES"

bigtwin, 2021-06-29:
I just noticed that _MAC OSX Big Sur_ requires a _Mac Mini_ from 2020! 
All former MAC OSX are fine with _Mac Mini_ 2014 and even older.

This may underline the importance of that issue. In order to now run _Companion_  on a MAC, you need to own the very latest (and expensive) Hardware. At least concerning a Mac Mini. 

This build-bug is really annoying :-(

kilrah, 2021-06-29:
That's very incorrect.

https://support.apple.com/en-us/HT211238

```Mac mini
Learn how to identify your Mac mini.

Mac mini (M1, 2020)
Mac mini (2018)
Mac mini (Late 2014)

MacFlieger, 2021-06-29:
The Mac itself is not the problem, cause the nightlies are working.

Would it be possible, to compile the release-version of companion with the nightlies-build-script?

bigtwin, 2021-06-29:
Ok, I was really wrong concerning the Hardware PreReqs of MacOSX. Thanks for this correction!
:-)

However the bug is still annoying. I am not able to migrate to _Big Sur_ and I really look forward to have back an official realease running on _Mojave_ (and surely on _High Sierra_).

kilrah, 2021-06-29:
Fixed, you can redownload release.

closed 2021-06-29 by kilrah

bigtwin, 2021-06-30:
I can confirm - _Companion_ official release 2.3.13 now runs on _MAC OSX Mojave_.
:-)
Thx for taing care of OTX!

MacFlieger, 2021-06-30:
Yes, it also works with High Sierra.
Thanks for your work.

Looking at the commits, the problem really was the additional "rm CMakeCache.txt" in the release-script.