opentx/opentx #8562: Companion 2.3.12: When choosing a sensor to include in a "Calculated" sensor, the minus signs are missing.

Miami-Mike, 2021-06-20:
In previous versions of Companion, when selecting a sensor to include in a "Calculated" sensor, each available sensor's name appeared in the top half of the list preceded by a minus (-) sign and again in the bottom half of the list without a preceding sign. Now the sensor's name still appears in both halves of the list but the minus sign is missing, making it unclear whether the value or its negative will be used in the calculation.

This is the issue that [sandet01 mentioned on RCGroups](https://www.rcgroups.com/forums/showthread.php?p=47330411&postcount=5536).

3djc, 2021-06-20:
Yeap, replicated. @elecpower could you have al look ?

elecpower, 2021-06-20:
Yes. Without looking I know where the bug is but I'm away this week so will put in a PR next week.

PR #8580 (2021-07-06): Fixes 8562 missing negative sign for telemetry sources
Fixes #8562
closed 2021-07-06 by 3djc