opentx/opentx #8565: Horus X10S Express and OTX 2.3.12 causes on some receivers huge delay in signals and causes malfunction on S-Bus and slow down in PWM Outputs

BernhardFe, 2021-06-21:
I updated my FRSky Horus X10S Express form OTX 2.3.11 to 2.3.12.

After the update some models working with FrSky Archer R8 Pro (FW2.1.7)and FrSky RSX-R (FW2.1.0) (both on ACCESS protocol) doesn't work with S-Bus to a Spirit FBL unit any more. 
Looking deeper I saw as well that there is a huge delay on signal processing (0,5 >2s) on the Archer R8 Pro to the PWM output with OTX 2.3.12. The PWM's of the Archer R8 Pro have nothing to do with the spirit unit. So it seems to be an issue between Horus and receiver.
This happend not to an Archer RS (FW2.1.8). This unit works fine together with the Spirit 

Roll back and forth between 2.3.11 and 2.3.12 let me reproduce the errors.

I expected that the Archer R8 Pro and the RSX-R work after the update to the new OTX version as before.

[image]
[image]

BernhardFe, 2021-06-21:
Update: Others found that the issue happens if 24 channels selected in the protocol to the receiver. If the channels reduced to 16 the issue seems to be fixed in 2.3.12. In 2.3.11. 24 channels are working as expected

Fridolin1981, 2021-06-21:
I had a similar issue:

Horus X10S Express, FW 2.1.6, OTX 2.3.12 together with the Archer SR8Pro FW 2.1.7. With OTX there seems to be an issue in the PWM transmission. I have flaps using the slow function. There is a delay of about 1.5 seconds before the servos start to move and stop in midposition. After another 1.5seconds the movement continues. 
After switching back to 2.3.11 there is no problem to be found.

BernhardFe, 2021-06-21:
If you need only 8 or 16 channels you can switch in the Horus at the receiver registration to 8 or 16 channels. Then you can use the OTX 2.3.12 and the signal is faster transmitted as well.

3djc, 2021-06-22:
Could you try with this firmware and confirml it does fix it ?

[x10express.zip](https://github.com/opentx/opentx/files/6694624/x10express.zip)

BernhardFe, 2021-06-23:
First test with Archer RS and Horus X10S Express, 16 CH mode, S-Bus connection to the spirit2 unit fails. Communication doesn't work Spirit2 says. Flashing back the Horus works again.

3djc, 2021-06-23:
Very true, another test if you do not mind

[x10express.zip](https://github.com/opentx/opentx/files/6699706/x10express.zip)

BernhardFe, 2021-06-23:
Test with Archer RS and Horus X10S Express, 16 CH mode and 24 CH mode, S-Bus connection to the spirit2 unit has been successful. No issues found any more.
Not tested: Other Archers and 8CH mode

3djc, 2021-06-23:
Thanks for your test and report !

BernhardFe, 2021-06-24:
Is this fix save to fly?

3djc, 2021-06-24:
yes

closed 2021-06-24 by bsongis