opentx/opentx #8566: Connecting USB on bootloader lock whole X9D+ radio

prikrylm, 2021-06-21:
**Describe the bug**
Switch on X9D+ (pre 2019) on bootloader (v2.3.12) and connect USB cable lock whole radio.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on radio in bootloader mode
2. plug in USB cable (USB connected appears on the screen)
3. No disk on PC appears
4. unplug USB cable - no menu appears
5. try to switch off radio - nothing happened, the radio still powered on "USB connected" is still on the screen.
6. only way  to "switch off" is unplug main battery.

**General information**
 - OpenTX version 2.3.12 
 - Hardware Frsky X9D+ (pre 2019)

kilrah, 2021-06-24:
Can't reproduce. Bad flash?

3djc, 2021-06-24:
Make sure you reflash your bootloader not from computer, but directly on the radio itself (using sd manager, click on firmware file, choose flash bootloader)

prikrylm, 2021-06-25:
Bootloader is flashed from radio. But probably is something wrong with my radio. Bootloader 2.3.11 behaves the same.

closed 2021-06-25 by prikrylm