opentx/opentx #8567: No more sound with version 2.3.12

Cliffweb, 2021-06-22:
**Describe the bug**
No more sound or alerts with this version 2.3.12 of Companion 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to every where you have defined sound or alert with special functions.
2. No sound...

I tested the previous version 2.3.11 with the same plane, abd same SD card -> it's work well.
It's work well if you manually push the play button in special functions tab.

**Expected behavior**
Sounds like in previous version 2.3.12

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version 2.3.12
 - X10s
 - OpenTX Companion 2.3.12 Mac OS 11.4

Cliffweb, 2021-07-15:
I reopen this issue because it still not resolved, included in last night compilation.

3djc, 2021-07-15:
duplicate #8591

closed 2021-07-15 by 3djc