opentx/opentx #8573: variable haptic feedback on transmitter, using Lua?

zortrop, 2021-06-25:
Is there a way to enable support for variable haptic intensity in Lua? 

Maybe in the playHaptic function, or using a new one?

I'd like to try mapping accelerometer telemetry to variable haptic feedback. Would be interesting to feel the forces the model is subjected to. 

There are probably other uses for this, as well...

Currently using a X-Lite S.

Thanks!

zortrop, 2021-06-29:
I tried this out by changing playHaptic to take an "intensity" argument. Then wrote a Lua script which uses that and the PLAY_NOW flag, so it's constantly updating. 

It works well! I recommend trying it with any type of aircraft. I'm currently testing with a micro quad, NewBeeDrone BeeBrain Pro - it's the only thing I have with accel telemetry...

There seems to be a bit of latency. I'll try to measure it, but I don't know if I can capture back to the flight controller or sensor.

shiroyagi, 2021-12-30:
Sounds like it would be good as an alternative to the vario tones.