opentx/opentx #8574: Taranis x9+2019 with S8R  Rx 2.3.13 software - jerky movement

Smoothead, 2021-06-25:
Taranis x9+2019  with S8R  receivers - ACCST D16
2.3.13 update (with 2.38 SD) tried update twice, same result. 

Tested on two models, again same problem on both.
Looks like poor comms - erratic jerky movement of control surfaces. 
At first I thought it was heavy interference in house, tried turning off PCs, router, Tv, still problems.

Went back to 2.35 and controls once again smooth and responsive on both models.

Am I missing something or is this a bug?

Help appreciated.

Smoothead, 2021-06-26:
Update - obviously there are many variables involved.
I have a large number of models and tend to use fairly complex programs.
I now have two SDs   with  2.35 + appropriate firmware &  2.38+2.3.13 firmware so I can reasonable easily swap between the two to try to find the issue.
Todays simplification is that I deleted most of the models leaving just 2 + 3 buddy box options. These I have attached.
When time allows I will try a simple program with an X8R receiver.   The file attached contains two models for the S8R receiver as described in yesterday's post - Model 19 ASW28   and   Model 25 Bixler (.otx in .zip)
[TwoModels.zip](https://github.com/opentx/opentx/files/6719794/TwoModels.zip)

EmanuelStassar, 2021-06-26:
I have the same issue I believe with my X12s with ISRM upgrade. I opened a bug report for this. Then I found yours with a similar problem. Its D16 protocol related on ACCESS capable transmitters it seems. Do you have any ACCESS receivers you can test?

mpjf01, 2021-06-27:
@Smoothead. I loaded the ASW28 model into my X9DP2019 Tx bound to a S8R receiver. I do see an element of "jerkiness" in the servo movement. I do not see this with the other models on my Tx - even when using the same receiver. Can I suggest that you create a simple 4Ch model and see if you experience these issues with that. If you don't, the effect is being caused by your model setup rather than an OpenTx/RF issue. I don't have the time to spend trying to unpick what seems to be a very complex model and I'm not sure how anyone could debug it easily. If you need a hand building a suitable model for a 6 servo glider have a look at Mike Shellims templates(RC-Soar). 
@EmanuelStassar. Are you using 2.3.13? The issue you are seeing was a "feature" of 2.3.12 in some cases and is supposed to have been fixed by the 2.3.13 release.

Smoothead, 2021-06-27:
Hello Emanuel - I don't have any ACCESS receivers - already have about 20  X8Rs, S8Rs, S6Rs + lots of early FRSKY Rxs. I don't like not being able to return to D16 if things go pear shaped with ACCESS!

Hello mpjf101, thank you very much for taking the trouble to load the ASW28.  Yes 2.3.13. I have tried a simple straight in - out model and don't see the jerkiness.  I also tried stripping out all the 'additional mixes' and playtrack sounds to get the file size down but it is still jerky.  Also the flapless Bixler program exhibits the same very slow response jerkyness.   Virtually the same program has been used on 4 of my models for about two years on an old 9D Tx and two friends who have 9D+  and X9 lite Txs with no problems - very quick and smooth.
The very similar programming has been running in the Bixler on the 9D+ 2019 with an S6R with no hint of jerkiness until I installed 2.3.13.  
I will continue to experiment for the benefit of the community as clearly 2.3.13 is a potentially DANGEROUS version.  Obviously I will not be flying with it, fortunately I can use my  old faithful 9D (2013) to fly the same programs with no slow responses.

Smoothead, 2021-06-27:
Ooops accidentally closed - reopening.

EmanuelStassar, 2021-06-27:
I think my problem is similar, but not identical. I have a vey simple model and its very choppy. see: https://youtu.be/TeWQ3LObI7c

Smoothead, 2021-06-27:
Thanks for video.  Yes similar but not identical, yours is jerky but at least consistent, mine is very much like being on the edge of range with updates more random - totally unflyable. 
No external module on the 9D+ 2019 Tx.  
Continuing with experiments - feedback from others appreciated.
The reason for some of my program's complexity, even on the simpler Bexler model is to make full use of the stability provided by the S8R and S6R. Normally I fly with light stabilisation on, but self level off, in Normal and Acro modes (SE) . The motor is only active in the upper 75% (approx) of stick travel, if you move below this the sound announces motor stopped. If you pull back further the gyros are activated and bleeps warn - self level. This brings a whole new level of confidence when practising stunt routines - no grasping for that 'er which was it' panic switch!
The (SE) low rates mode is used for initial training and FPV - gyro stability permanently on.
All good fun.

Smoothead, 2021-06-29:
Since we have two parallel threads about the 2.3.13 jerky control please see thread open by Emanuel ---
Horus X12 - ISRM Open-Tx 2.3.13 - Very choppy servo movement on ACCST D16 receivers.

kilrah, 2021-07-02:
Duplicate of #8575

closed 2021-07-02 by kilrah