opentx/opentx #8575: Horus X12 - ISRM Open-Tx 2.3.13 - Very choppy servo movement on ACCST D16 receivers

EmanuelStassar, 2021-06-26:
After updating my X12s to 2.3.13 the servo movement became very 'choppy' in ACCST D16 mode. A clubmate with the same Tx config (X12s with ISRM upgrade) has the same issue. On ACCESS all is fine.

I switched the D16 models I had on the field today to the MPM and there control was fine. 

I went from 2.3.11 to 2.3.13 so I don't know if 2.3.12 had the same issue.

The way it is now is unflyable so this is a serious bug!

Smoothead, 2021-06-27:
Obviously we have the same problem Emanuel.  It is suggested in my thread that it is the complexity of my model program - perhaps you could submit the program(s) you are seeing the problems with - any commonality may give the open tx coders a clue.
On the subject of the coders - thanks for all your brilliant efforts guys I have been a fan of OpenTx for 9 years and now have most of my club converted.

EmanuelStassar, 2021-06-27:
I made a short video to demonstrate how severe the issue is. I put a Slow on the rudder so I could clearly show the difference.
https://youtu.be/TeWQ3LObI7c
To clarify, in the first section of the video the X8R is bound to the ISRM module. During the second part its bound to the MPM. Nothing else was changed.

jetpathar, 2021-06-28:
I have the same problem, very serious issue although the release mention 'Fix Horus ISRM jittery pulses'.
X12s OTX2.3.13 ACCST-D16 XR8R-Pro
You can check my [SNOOP65.CSV.txt](https://github.com/opentx/opentx/files/6726036/SNOOP65.CSV.txt) log file (using Companion) of the S-Bus output made with a Snoop S-bus Data Logger.
On channels 1, 2 & 4 (ailerons, elevator & rudder) I'm moving the sticks gently from one end to the other.
Channel 8 shows a triangle signal from OpenTX: L01: a<x; ch8; 0 / Ch8 mixer: L01; weight  100; delay up 1.0; delay down 1.0; slow up 2.0; slow down 2.0
PWM Width shows the actual servo output on channel 8
The problem is obvious, it seems that the channels are sampled at a lower frequency than they should be.
I'm returning to OTX2.3.11

raphaelcoeffic, 2021-06-28:
> The problem is obvious, it seems that the channel signals are generated or sampled at a lower frequency than they should be.

that would be obvious if FrSky would publish what the module expects. Instead, everyone has to reverse engineer their module, with no guaranty it won’t change with the next firmware.

3djc, 2021-06-28:
We did fix the module when using access receivers (and the fix works afaik), but now it pops with ACCST receivers. Why on earth does it work differently between the two is beyond me. But as we severely lack access equipment, starting with radios, it is very difficult for us to test or troubleshoot those

jetpathar, 2021-06-28:
Now the problem for me is:
How can I return to 2.3.11 since I have installed the ISRM module at the same time of upgrading to 2.3.13
Companion does not allow to compile/download an older version, I still have my original 2.3.11 image
but without the internalaccess build option therefore only for the older IXJT module.

Would someone have an image of 2.3.11 for X12s with the settings: english, ppmus, lua, internalaccess, Mode-2, AETR ?

EmanuelStassar, 2021-06-28:
I have almost the same config file. I dont have ppmus but the rest is the same. How can I get it to you?

jetpathar, 2021-06-28:
Hello Emanuel,
Thank you so much, I guess you can attach the file with the option below the box appearing when you write the post.
You would need to add another extension like .txt otherwyse it is not accepted.

EmanuelStassar, 2021-06-28:
[opentx-x12s-internalaccess-lua-en-2.3.11-otx-1.bin.txt](https://github.com/opentx/opentx/files/6728425/opentx-x12s-internalaccess-lua-en-2.3.11-otx-1.bin.txt)
Here you go!

mpjf01, 2021-06-28:
> We did fix the module when using access receivers (and the fix works afaik), but now it pops with ACCST receivers. Why on earth does it work differently between the two is beyond me. But as we [...]

I have a number of transmitter/receiver combinations that I could use to help out if need be. Others here the same no doubt. These include X12S ISRM, X10S ISRM, X9DP2019SE, X9DP, X-lite and D16 and ARCHER receivers in my case.

jetpathar, 2021-06-29:
Hey Emanuel, flashed your image and looks good so far, thank you !

Smoothead, 2021-06-29:
See my previous comments in Taranis x9+2019 (EU version) with S8R Rx 2.3.13 software - jerky movement #8574
Have just been testing 2,2.13  (FCC version)on my old 9D Tx  running the same - no problems with that.
So not sure whether the issue is caused by the EU / FFC difference or the different chipsets in the 9D+ 2019 and 9D?
In both cases I am using the internal modules (although I also have an IRX4 module in the 9D).

Smoothead, 2021-06-29:
Re 2.2.13 in Tx 9D+2019(EU) with SR8 Rx.   Updated Rx firmware to 2.10 and Tx  ISRM 2.16.  This allowed me to run 2.2.13 with the less complex model programs but still jerky with the more complex.   Tried 2.2.12 and all models silky smooth. Hope this helps someone with the problem solving.

brainbubblersbest, 2021-07-02:
Doesnt need the isrm the heartbeat Wenn running in accst Mode?
Im asking this when the externalaccessmod was introduced.
Thats why i preferred the solution with the external Inverters. The ixjt uses the heartbeat. And I recognized the Code changes that where made in

raphaelcoeffic, 2021-07-02:
> Doesnt need the isrm the heartbeat Wenn running in accst Mode?

It does. The code sets a timer with the mixer scheduler, which is only their as a "security" (expected period plus some head room for jitter), and the heartbeat detector uses the same flag to trigger the mixer. So it should be checked the mixer is not triggered twice (once with heartbeat, and once with the security timer), but that would be all.

robustini, 2021-07-02:
@EmanuelStassar could you try this firmware and give me feedback?
It is compiled to a specific [commit](https://github.com/opentx/opentx/commit/a167ded9b534f373904e4424e053187b90237b8a), let me know if there is the problem or not.

[firmware.zip](https://github.com/opentx/opentx/files/6756136/firmware.zip)

3djc, 2021-07-02:
Of this this commit is fine on ACCST, so it will work for EmanuelStassar but it does not for ACCESS. We have a very clear idea of what the issue is, but we lack equipment and time to fix it, thats all

raphaelcoeffic, 2021-07-02:
> Of course this this commit is fine on ACCST, so it will work for EmanuelStassar but it does not for ACCESS. We have a very clear idea of what the issue is, but we lack equipment and time to fix it, thats all

I should be able to check that as soon as I’m back from vacations. So in about 2 and a half weeks.

robustini, 2021-07-02:
@3djc you know what the offending commit is...  ;-)

EmanuelStassar, 2021-07-02:
I will try you firmware tomorrow. Its absolutely safe to flash to Horus X12s? I will check both ACCST and ACCESS receivers.

robustini, 2021-07-03:
> 
> 
> I will try you firmware tomorrow. Its absolutely safe to flash to Horus X12s? I will check both ACCST and ACCESS receivers.

Obviously yes.
You can flash a completely corrupt firmware, including the bootloader, and there will always be the possibility to restore the tx with a correct firmware, the STM32 bootloader will never corrupt it with a bad flash.
With this compiled version you will see that there will be no more jerky movements.

EmanuelStassar, 2021-07-03:
Just tested your firmware and it works fine. I don't see any difference in servo sersponse with version 2.3.11. 
What has changed?

robustini, 2021-07-03:
If you read my issue that I posted above you will understand.
Unfortunately, massive changes have been introduced since a certain commit, and something has compromised the proper functioning of the ISRM.

robustini, 2021-07-03:
It could be a bug inside the ISRM module firmware, but FrSky obviously doesn't release the sources, so for OpenTX developers this is a big limitation.
But one thing is little but sure: before that commit everything worked.

Smoothead, 2021-07-03:
In view of all the above can anyone advise on the **safest** versions of Open Tx & ISRM to use with 9D+SE 2019 Tx and ACCST 16 SR8 receivers?  Difficult to know how many versions to rewind (currently on 2.3.12, ISRM 2.16. Rx firmware to 2.10 and Tx ISRM 2.16, which seems OK re jitter but getting concerned that a **nasty** will appear during a flight.  Any help appreciated.

robustini, 2021-07-03:
2.3.11 is the safe at the moment, or 2.3.12 V219 compiled at [that commit](https://github.com/opentx/opentx/commit/a167ded9b534f373904e4424e053187b90237b8a).

goldendark, 2021-07-03:
I tried robustini firmware. It works well but the radio is claiming for 2.3V0036 sdcard.
thx robustini

goldendark, 2021-07-03:
> 2.3.11 is the safe at the moment, or 2.3.12 V219 compiled at [that commit](https://github.com/opentx/opentx/commit/a167ded9b534f373904e4424e053187b90237b8a).

Hi I went to your link but how to download the 2.3.12 V219?

robustini, 2021-07-03:
> 
> 
> I tried robustini firmware. It works well but the radio is claiming for 2.3V0036 sdcard.
> thx robustini

https://downloads.open-tx.org/2.3/nightlies/sdcard/

goldendark, 2021-07-03:
> > I tried robustini firmware. It works well but the radio is claiming for 2.3V0036 sdcard.
> > thx robustini
> 
> https://downloads.open-tx.org/2.3/nightlies/sdcard/

thx

Smoothead, 2021-07-03:
Many thanks guys.

Gewe99, 2021-07-04:
Hello 
I have the sticky servo movement with Horus X12S and original IXJT Module Accst D16 with R-XSR receiver. OTX 2.3.13
See also:
https://frsky-forum.de/index.php?thread/5651-opentx-2-3-13-update/&postID=73986#post73986
Hope the ISRM Fix works also for the IXJT...

3djc, 2021-07-07:
Could you try  those ?
[test-firmwares.zip](https://github.com/opentx/opentx/files/6774751/test-firmwares.zip)

Smoothead, 2021-07-07:
Thanks 3djc - I have just tried the 9D+2019 version in your zip file and the responses are fast and smooth even on very complex model setup - looks good. I hope this works for other Txs too.    
On the subject of testing other versions, in an earlier post I was advised " 2.3.11 is the safe at the moment, or 2.3.12 V219" but can't find either on the site, if I re-install the Companion 2.3.11 it sees 2.3.13. Apologies for my ignorance!  Perhaps this test file is my best bet?

3djc, 2021-07-07:
You cannot go back to old version if you did not keep the firmware file. Stay on this test version for now

Smoothead, 2021-07-07:
Unfortunately upgraded from 2.3.5 originally. 
Obviously I need to log on more regularly and keep all firmwares in future!
Will give the your file a thorough test before flying and let you know if I spot any issues.
Thanks for your help, much appreciated.

3djc, 2021-07-07:
New set

[test-firmwares.zip](https://github.com/opentx/opentx/files/6775950/test-firmwares.zip)

Smoothead, 2021-07-07:
Well 'New set' also seems OK on 9D+2019 (EU)

PR #8601 (2021-07-08): More pulse sync fixes
Fix Horus iXJT where HB and expired timers where both trigering

Fix ISRM ACCST broken by 2.3.13. Duration set to 10ms which is the longuest one (9ms for ISRM/ACCST) + 1ms backup

This fixes #8585 
This fixes #8575
closed 2021-07-08 by bsongis