opentx/opentx #8576: Bug OpenTx 2.3.12 et 2.3.13 with mutiprotocole module Jumper 4 in 1

sjorger, 2021-06-26:
J'utilise un module multiprotocole "Jumper 4 in 1"

Il fonctionne tout à fait normalement sous OpenTx 2.3.11 mais pas avec les versions supérieures.
Sous OpenTx 2.3.12 et 2.3.13, il est impossible de binder un récepteur car la sélection des protocoles se freeze instantanément lorsque l'on choisit l'option "Multi".

Je précise que j'ai le même problème sur deux radios différentes (Horus X12S).

3djc, 2021-06-26:
English only please

sjorger, 2021-06-26:
I am using a "Jumper 4 in 1" multiprotocol module

It works quite normally under OpenTx 2.3.11 but not with higher versions.
Under OpenTx 2.3.12 and 2.3.13, it is impossible to bind a receiver because the selection of protocols is instantly frozen when the "Multi" option is chosen.

I specify that I have the same problem on two different radios (Horus X12S).

closed 2021-06-26 by sjorger

RickzFPV, 2021-06-26:
That is really weird because I am using my Jumper 4 in 1 module on both my QX7 and my jumper T16s after updating and it works perfectly