opentx/opentx #8582: Internal GPS on RadioKing TX18S

ViorelX, 2021-06-29:
Hello,
I have got a RadioKing TX18S a couple months ago and I have managed to do the next hardware modiffications in order to improve it as much as possible:
1. 2s battery from 21700 5000mAh cells;
2. balancing connector for charging the battery;
3. Horus X10/12 Bluetooth module;
4. modiffication for allowing external R9M 2019 in ACCESS mode: https://github.com/opentx/opentx/wiki/Externalaccessmod;
5. 2 additional potentiometers (EX1, EX2) and 2 additional switches (SI, SJ): https://www.rcgroups.com/forums/showpost.php?p=44888401&postcount=5317
6. internal GPS with RX, TX wires soldered to 200, 201 pins of the processor, which are its UART8 RX, TX terminals (PE0, PE1).

[image]

[image]

SI is a toggle switch put on the bottom left corner:
[image]

I moved the 2.4GHz antenna in order to allow the external module antenna to be freely bent towards the user, so it can stay vertically, as normally needed. I put a toggle switch (SJ) in the original hole of the antenna.
[image]

I have flashed the latest 2.3.13 OpenTX firmware (for Jumper T18) and all is working fine, excepting the GPS, which is not implemented yet.
I have compiled by myself the sources of 2.3.13 OpenTX and it works, as well.

Next, in order to enable the internal GPS, I have modified so far only 2 source files:
-https://github.com/opentx/opentx/blob/release/2.3.13/radio/src/targets/horus/CMakeLists.txt
-https://github.com/opentx/opentx/blob/release/2.3.13/radio/src/targets/horus/hal.h

I managed to compile the project with these 2 modified source files, but it doesn't work on the sender. The screen becomes white and the sender gets blocked.

What have I changed on hal.h:
#if defined(RADIO_T18) && defined(INTERNAL_GPS)
#define GPS_RCC_AHB1Periph            RCC_AHB1Periph_GPIOE
#define GPS_RCC_APB1Periph            RCC_APB1Periph_UART8
#define GPS_RCC_APB2Periph            0
#define GPS_USART                     UART8
#define GPS_GPIO_AF                   GPIO_AF_UART8
#define GPS_USART_IRQn                UART8_IRQn
#define GPS_USART_IRQHandler          UART8_IRQHandler
#define GPS_UART_GPIO                 GPIOE
#define GPS_TX_GPIO_PIN               GPIO_Pin_1 // PE.01
#define GPS_RX_GPIO_PIN               GPIO_Pin_0 // PE.00
#define GPS_TX_GPIO_PinSource         GPIO_PinSource1
#define GPS_RX_GPIO_PinSource         GPIO_PinSource0
#elif !defined(INTERNAL_GPS)
  #define GPS_RCC_AHB1Periph            0
  #define GPS_RCC_APB1Periph            0
  #define GPS_RCC_APB2Periph            0
#endif

These lines are similar to the original source lines for enabling internal GPS on Horus X12S:
// GPS
#if defined(PCBX12S)
  #define GPS_RCC_AHB1Periph            RCC_AHB1Periph_GPIOA
  #define GPS_RCC_APB1Periph            RCC_APB1Periph_UART4
  #define GPS_RCC_APB2Periph            0
  #define GPS_USART                     UART4
  #define GPS_GPIO_AF                   GPIO_AF_UART4
  #define GPS_USART_IRQn                UART4_IRQn
  #define GPS_USART_IRQHandler          UART4_IRQHandler
  #define GPS_UART_GPIO                 GPIOA
  #define GPS_TX_GPIO_PIN               GPIO_Pin_0 // PA.00
  #define GPS_RX_GPIO_PIN               GPIO_Pin_1 // PA.01
  #define GPS_TX_GPIO_PinSource         GPIO_PinSource0
  #define GPS_RX_GPIO_PinSource         GPIO_PinSource1
#elif !defined(INTERNAL_GPS)
  #define GPS_RCC_AHB1Periph            0
  #define GPS_RCC_APB1Periph            0
  #define GPS_RCC_APB2Periph            0
#endif

I only changed the port UART4 to UART8, which is seemingly free.
Why doesn't work this modiffication? What should be done?

3djc, 2021-06-29:
This radio is not supported

closed 2021-06-29 by 3djc

ViorelX, 2021-06-29:
What means 'This radio is not supported'?
It works very well with firmware for Jumper T18. I want to add the internal GPS, as allready done for RadioMaster TX16S.

kilrah, 2021-06-29:
We never had contact with the manufacturer of this radio, we don't have one, so we aren't going to spend time trying to figure things out for it.

ViorelX, 2021-06-29:
OK, but it is fully compatible to Jumper T18. Considere the issue is related to Jumper T18.

ViorelX, 2021-06-29:
I have modified my post: the topic is now about the supported Jumper T18, with the mention I actually own a RK TX18S.

kilrah, 2021-07-01:
We know there are differences since people have reported not everything working correctly with a T18 build in a TX18S, but we don't know what changes.

svpetar, 2022-01-07:
When you did all of this, get EDGETX and enable your touch screen. It's already there and motherboard is ready.

svpetar, 2022-05-15:
Hi! I added additional four channels, but two proportional ones does not have full travel. What kind of pots are you using? I did it with 5k pots.

kilrah, 2022-05-15:
That's what you typically get if you configured the pot as being "pot with detent" in the hardware menu but did not center it during the first calibration step.

ViorelX, 2022-05-15:
I have used regular 5k pots too and work fine, with full travel of 1000-2000us.

svpetar, 2022-05-15:
Thanks! I just realized the same.