opentx/opentx #8584: Some field values are not translated into german language (since opentx-version 2.3.11)

Redmilan, 2021-06-30:
In the Companion (since version 2.3.11), despite the selected setting "German" for the language settings, "English terms" are mixed into certain fields on almost all pages, e.g. on the "Mixer page" the term "weight" is given instead of "Gewicht" and instead of "Schalter" the term "switch". 

This is only one example; on almost all pages you will find this mix of English expressions with German language settings (denglish). In Opentx version 2.3.10 this confusion was not yet present. If you change the language settings to "French", this problem is not noticeable (I would have been surprised!). However, the problem may also occur with other language settings.

It would be nice if the problem could be fixed in future versions of opentx.

3djc, 2021-06-30:
If you do not like the German translation, they best is to head over https://github.com/opentx/opentx/wiki/Flow-for-the-Translators and start improving those

closed 2021-06-30 by 3djc

Redmilan, 2021-06-30:
I feel a little misunderstood. It's not about translating English expressions better in the Companion. The translations in opentx version 2.3.10 are perfectly fine. Only from version 2.3.11 to 2.3.13 errors have crept in there. I suppose that the translations for the different languages are stored in a database. In the Companion these translations are then called via the corresponding database reference. Apparently these references for some input/output fields or selection fields are no longer correct when "German" is selected as language.Instead of the reference for the German expression, the reference for the English expression is apparently used in the affected fields. Your suggestion to revise the translations is what I would call a back pass in football.

After all, the translations were perfectly fine in Opentx version 2.3.10. My point was only to inform that from version 2.3.11 on this is no longer so.

3djc, 2021-06-30:
It just means those screen have envolve in english, and a translator need to review those again, there is not much non german speaker can do to improve

Redmilan, 2021-07-01:
First of all, thank you very much for replying to my comment. However, the question still remains for me why opentx version 2.3.10 and the previous versions are largely error-free and the problem only starts with version 2.3.11. I can't imagine that for each new release of 2.3 such translation work has to be done. In this case, a redesign of this part of the application should be considered.

However, the whole thing would be understandable to me if versions 2.3.11 to 2.3.13 were pre-releases of the upcoming version 2.4. 

Finally, thank you again for answering my questions.

3djc, 2021-07-01:
Each time screen are redesigned, translation works need to be done, and thanks to Elecpower, quite a few screens have been improved, and no 2.3.11 to 2.3.13 are not at all pre releases of 2.4