opentx/opentx #8585: LOST CONNECTION OF HORUS X10S EXPRESS AND RECEIVER

afdofdez, 2021-06-30:
I have a Horus X10S Express and both in versions 3.3.12 and 3.3.13 (much less) I have had problems of loss of connection with the plane, the Horus X10S said "loss of telemetry" and the servos stopped during one or two seconds, then the telemetry would recover and the servos would work again.
To produce the fault, he moved the right stick continuously, commanding ailerons and depth.

Both the Horus X10S Express transmitter with the receivers (Archer R6 and Rx6R) are updated to the latest versions of Frsky, namely 2.1.6 and 2.1.7. respectively.

I am back to version 3.3.11 and everything seems to work fine.

3djc, 2021-06-30:
ACCESS issue was know in 2.3.12, but we have several tester with very complex setup that said ACCESS was ok, but now in 2.3.13, it seem ACCST on ISRM is having issue. Are you saying you still have issues with ACCESS ?

afdofdez, 2021-06-30:
Yes, my problem is with Access Enviado desde Correo para Windows 10 De: 3djcEnviado: miércoles, 30 de junio de 2021 19:59Para: opentx/opentxCC: Fernando Fernández; AuthorAsunto: Re: [opentx/opentx] LOST CONNECTION OF HORUS X10S EXPRESS AND RECEIVER (#8585) ACCESS issue was know in 2.3.12, but we have several tester with very complex setup that said ACCESS was ok, but now in 2.3.13, it seem ACCST on ISRM is having issue. Are you saying you still have issues with ACCESS ?

3djc, 2021-06-30:
Then it would be nice to have a otx of your setup where this appears the more clearly

afdofdez, 2021-06-30:
[image]

[image]
[image]

afdofdez, 2021-07-01:
wisel and extra 300L models

[modelos.zip](https://github.com/opentx/opentx/files/6745904/modelos.zip)

brainbubblersbest, 2021-07-02:
Update to 2.3.13. Youre on 2.3.12

afdofdez, 2021-07-02:
I am now on 3.3.11
The problem occurred frequently in 3.3.12 and less frequently in 3.3.13, so I went back to 3.3.11.

bigtwin, 2021-07-02:
You continue to type 3.3.x instead of 2.3.x. Any special reason for that? May some misunderstanding ...

afdofdez, 2021-07-02:
just a typo.

robustini, 2021-07-02:
@afdofdez could you try this firmware and give me feedback?
It is compiled to a specific [commit](a167ded), let me know if there is the problem or not.
I compiled it for you for Spanish language.

[firmware.zip](https://github.com/opentx/opentx/files/6756107/firmware.zip)

afdofdez, 2021-07-02:
Do I install it directly from Companion?

robustini, 2021-07-02:
Thanks for the test!
@3djc  knows where to look...

raphaelcoeffic, 2021-07-06:
> Thanks for the test!
> @3djc knows where to look...

@robustini we already know there is an issue with the new mixer trigger ;-) there is no point to prove here. It would be much more useful to find the issue itself.

robustini, 2021-07-06:
I absolutely don't want to blame anyone, I just want to point out where the problem lies.

robustini, 2021-07-06:
> 
> 
> > Hi Raphael! I absolutely don't want to blame anyone, I just want to point out where the problem lies.
> 
> Ok ok! Anyway, I’ll take care as soon as I can. I’m currently away in the mountains, which makes it hard to debug this ;-)

Have a nice holiday!

3djc, 2021-07-07:
Could you try those ?
[test-firmwares.zip](https://github.com/opentx/opentx/files/6774746/test-firmwares.zip)

afdofdez, 2021-07-07:
Now I'm on vacation, this weekend I'll do the tests, ok?

afdofdez, 2021-07-08:
The version 2.3.13 variant 219 from 2021-07-05 08:20:03 that you have sent me works perfectly in the two planes

3djc, 2021-07-08:
Unfortunatly, this test is useless and does not give any info moving forward. Only usefull test are the version I posted.

afdofdez, 2021-07-08:
This is the version you posted yesterday.
But in Companion it says it's from 2021/7/5
[image]

3djc, 2021-07-08:
Great then ! Thanks for the feedback, looks like we have a fix for this issue :)

closed 2021-07-08 by bsongis

i3dm, 2021-08-19:
if there is a radio connection issue shouldn't this be a critical bug and message to all users to upgrade fw?

3djc, 2021-08-19:
Since you cannot actually  start your craft, you'll notice quicky, and fix is in 2.3.14 which is flagged as a should do upgrade

i3dm, 2021-08-19:
> Since you cannot actually start your craft, you'll notice quicky, and fix is in 2.3.14 which is flagged as a should do upgrade

why cant you start your craft?

PR #8601 (2021-07-08): More pulse sync fixes
Fix Horus iXJT where HB and expired timers where both trigering

Fix ISRM ACCST broken by 2.3.13. Duration set to 10ms which is the longuest one (9ms for ISRM/ACCST) + 1ms backup

This fixes #8585 
This fixes #8575