opentx/opentx #8587: Downgrade option

Btje1977, 2021-07-03:
When downloading a previous companion version it is intended to download this version to the transmitter.
I have the problem of a fresh start and the 2.3.13 is giving jitter on the servo. I'm not able to go back to the previous version as
i don't have a previous version.

So when downloading companion 2.3.11 the user does not want to upgrade to 2.3.13 he want's to use this verion.

3djc, 2021-07-03:
Firmware are compiled on demand, and are always supplied last version. If you want to be able to go back, simply keep a copy of the binaries you are using.

closed 2021-07-03 by 3djc

Btje1977, 2021-07-03:
But this is wat I mention. If you start you don't have a binaries of a previous version.

3djc, 2021-07-03:
You can some older from radio manufacturers

bigtwin, 2021-07-03:
I also stumbled accross this fact. If I run Companion 2.3.11, I surely have the expectation that I get 2.3.11 to the transmitter. And I would not want to keep a binary with all kind of build options which I may need in the future or not. So, I would also appreciate to have it like this: Companion 2.3.xyz generates OTX 2.3.yxz.

However, I understand that this is obviously not possible. But how comes Companion is available in all obsolete Versions? Why all the obsolete SD-Card contents? In this context, it does not look very sensefull to me.

3djc, 2021-07-03:
Because there is 1 version of Companion 2.3.11, the same for everyone, and several thousand versions of firmware, compiled on demand

Btje1977, 2021-07-03:
Yes and that is a good thing but when 2.3.11 is downloaded. It should not do an update if check update is not used or the update is turned off. People sometimes do not want the latest version. Now  opentx is forcing them and this is not the right way. Is it compiled on my PC or is it getting something from a server?

bigtwin, 2021-07-03:
Please note, that I choosed 2.3.11 just as an sample to express what I mean. 
I should be like "_Why do we get all Companion Versions from 2.3, when OTX will always be compiled with the latest source?_"
But again, I can surely live with it. It simply does not look consequent to me.

kilrah, 2021-07-03:
Simply for practical reasons. Leaving an old companion version is just leaving it there, building old binaries would mean building infrastructure for it.