opentx/opentx #8588: Flat spots in BF BB Logs caused by late OpenTx pulses delivery

tonycake, 2021-07-03:
This is going to be a fun one... 

For some time now we have been noticing 'flat spots' in betaflight Black Box logs, which appeared to be due to duplicate packets on the Rx side. 

Instead, what appears to be happening is that synced protocols (GHST for example, but quite probably all of them) occasionally run the mixer late (1ms typically), and the transmitter module has no 'fresh' data to deliver OTA to the receiver. 

To see it happening, setup a logic analyzer/scope to monitor the serial data stream from the radio, and look for packets which are late by approx. 1ms. 

These occur at random times, with a period of 'a few seconds', and no particular pattern. 

Seems to happen on all radios, confirmed on the TX16s, and a XLite Pro (so not Color display related). 

The below scope screenshot shows the problem occurring. The cyan trace being the mixer scheduler timer interrupt, which correctly sets the mixerFlag. The magenta trace being the mixer actually running (mixerSchedulerWaitForTrigger completing without a timeout). 

Myself and 3djc have spent many hours on this so far, with no real progress, and a suspicion that the issue lies in the RTOS itself. Almost seems that the RTOS detects the flag being set, but doesn't correctly task switch. 

[image]

Some testing that has been performed so far: 
- Change CFG_SYSTICK_FREQ from 1ms to 2ms, mixer runs 2ms late when the problem occurs, as kindof expected
- Switch from using flags to semaphores to signal a mixer run, no change in behavior
- Shut down the audio task, no change

JBKingdon, 2021-07-04:
Great work Tony! Many thanks for investigating this and documenting the problem, it will benefit everyone if we can get to the bottom of it.

Allerhopp, 2021-07-05:
Probably not related, but we saw a similar effect watching a triangle signal to evaluate link quality. Did you test with ErskyTX?
The issue was discussed as "fang effect" here:

https://www.rcgroups.com/forums/showthread.php?3507071-Important-Firmware-Update-%C2%96-ACCST-D16-2-1-x/page177#post44379023

3djc, 2021-07-05:
This is unrelated @Allerhopp, this is really an issue with pulse sync sometimes (2 to 8 times per minute) shifting mixer run by one RTOS tick