opentx/opentx #8591: Audio files not working in simulator on macOS

Gernod, 2021-07-04:
Audio files are not working in simulator under macOS Big Sur

Steps to reproduce the behavior:
1. choose Model in companion
2. add play audio file to switch
3. start simulator
4. activating the switch does not play audio

 - OpenTX version 2.3.12
 - Hardware (Radiomaster T16S)
 - OpenTX Companion 2.3.12 and 2.3.13 on macOS Big Sur (problem does not exists on OpenTX 2.3.11 and earlier)

bigtwin, 2021-07-04:
Just checked: No issue with Companion 2.3.13 on Mac OSX Mojave, Profile QX7. everything works fine.

Gernod, 2021-07-04:
different mac with Big Sur, profile: X9-lite same problem. Seems to be a Big Sur specific problem.

Cliffweb, 2021-07-04:
+1 with Big Sur
See the issue #8567, now closed, because the Opentx team developers don't have Mac system, in order to check the MacOs issues and correct them.
Then if the issues are not reproductible on Win, there is no chance to see the MacOs issues fixed.

3djc, 2021-07-04:
And proven wrong by https://github.com/opentx/opentx/commit/eb56dbea914ebe6dfb41431878d847eb222e655a ….

bigtwin, 2021-07-04:
@Cliffweb - you refer to a closed issue in order to document that OTX developers dont have MACs and cannot track down errors. But this issue was closed by you, so it is not really clear to me what you wanna say. Also - we just had a [problem with MACs](https://github.com/opentx/opentx/issues/8561) on 2.3.12/13 which was resolved perfectly.
I guess we simply need to be a little patient and always provide close feedback and detailed descriptions about MAC issues.

Cliffweb, 2021-07-05:
> @Cliffweb - you refer to a closed issue in order to document that OTX developers dont have MACs and cannot track down errors. But this issue was closed by you, so it is not really clear to me what [...]

I have closed this issue, because without Mac system, the Opentx team cannot reproduce the problem and fix it.
OpenTx team put this MacOs version on the table, but at the first problem he answer "we cannot fix the problem because we havent Mac system"
Anyway, I have time, and the problem is minor, the system work fine on my radio, this is the most important
Bonne journée.

bigtwin, 2021-07-05:
So if it is not resolved and still relevant, I would recommend to re-open the issue.

BernhardFe, 2021-07-13:
I have the same issue and in the german FrSky Forum others documented the same issue. 2.3.11 is working, 2.3.13 is now on my iMac and the sound simulation is not working any more. Between both versions there was no update on my iMac. 
it seems there might be a compiling issue?

Cliffweb, 2021-07-15:
> So, if it is not resolved and still relevant, I would recommend to re-open the issue.

Ok I did it, it's reopened.

Cliffweb, 2021-07-15:
Ouha ! so fast ! 3djc close them again at the light speed !
We want the same speed for issue resolution !

BernhardFe, 2021-07-15:
Thanks a lot, otherwise I can not control in advance if the right audio files are assigned to a dedicated action.

timtuxworth, 2021-07-19:
I have the same problem. I just got started using Companion and never used the older versions, so I just assumed audio was never supposed to work in the simulator. Good to see it's just a bug that is being fixed.

Cliffweb, 2021-07-26:
Nothing new on 2.3.14
Sound still not working on simulation.

3djc, 2021-07-26:
Should not come as a surprise since this issue is still opened

Cliffweb, 2021-07-26:
Yes you're right, I still to believe Santa Claus will come to my chimney.
It's my blue flower side.
May the force be with you.

BernhardFe, 2021-08-12:
One hint may help: In the special function tab the audio playing works. Only in simulation it is not working. Playing the Audio file in general might not be the problem. For mit it looks like missing trigger in simulation. But looking in the debug log in simulation I can see the code for sounds:
playFile("/SOUNDS/de/GFrei.wav", flags=0, id=19)
playFile("/SOUNDS/de/MEin.wav", flags=0, id=1)
playFile("/SOUNDS/de/FStart.wav", flags=0, id=5)
So can it be a problem with the volume of the sound that is not recognised?

derFliegendeHamburger, 2021-10-06:
on macOS 11.6 last  version with working sound in simulator for me is  2.3.11.
the two debug outputs (cannot attach files here)

`2.3.11

StartAudioThread(11)
simuFatfsSetPaths(): simuSdDirectory: "/path/to/SD-copy"
simuFatfsSetPaths(): simuSettingsDirectory: ""
reusableBuffer: modelSel=333, moduleSetup=212, calib=110, sdManager=632, hardwareAndSettings=184, spectrumAnalyser=452, usb=512
opentxInit
f_close(0x0) (FIL:0x114ed8338)
convertToSimuPath(): /SOUNDS/en/MyModel -> /path/to/SD-copy/SOUNDS/en/MyModel
f_opendir(/path/to/SD-copy/SOUNDS/en/MyModel) = error 2 (No such file or directory)
convertToSimuPath(): /IMAGES/trex450.bmp -> /path/to/SD-copy/IMAGES/trex450.bmp
findTrueFileName(/path/to/SD-copy/IMAGES/trex450.bmp)
	found: /path/to/SD-copy/IMAGES/trex450.bmp
f_open(/path/to/SD-copy/IMAGES/trex450.bmp, 1) = 0x7fff8067a450 (FIL 0x70000980cc38)
f_size(0x7fff8067a450) 1144
f_close(0x7fff8067a450) (FIL:0x70000980cc38)
convertToSimuPath(): /SOUNDS/en/SYSTEM -> /path/to/SD-copy/SOUNDS/en/SYSTEM
f_opendir(/path/to/SD-copy/SOUNDS/en/SYSTEM) = OK
f_closedir(0x70000980cdb8)
opentxStart(7)
luaInit
playFile("/SOUNDS/en/SYSTEM/0000.wav", flags=0, id=67)
playFile("/SOUNDS/en/SYSTEM/volt1.wav", flags=0, id=67)
playFile("/SOUNDS/en/thrhld.wav", flags=0, id=5)
playFile("/SOUNDS/en/SYSTEM/0000.wav", flags=0, id=16)
playFile("/SOUNDS/en/SYSTEM/volt1.wav", flags=0, id=16)
lsScripts 0x7fc2b2e79710
convertToSimuPath(): /SOUNDS/en/SYSTEM/0000.wav -> /path/to/SD-copy/SOUNDS/en/SYSTEM/0000.wav
findTrueFileName(/path/to/SD-copy/SOUNDS/en/SYSTEM/0000.wav)
	found: /path/to/SD-copy/SOUNDS/en/SYSTEM/0000.wav
f_open(/path/to/SD-copy/SOUNDS/en/SYSTEM/0000.wav, 1) = 0x7fff8067a450 (FIL 0x114ed47c0)
convertToSimuPath(): /LOGS -> /path/to/SD-copy/LOGS
f_opendir(/path/to/SD-copy/LOGS) = OK
f_closedir(0x70000980cd70)
convertToSimuPath(): /LOGS/MyModel-2021-10-06.csv -> /path/to/SD-copy/LOGS/MyModel-2021-10-06.csv
findTrueFileName(/path/to/SD-copy/LOGS/MyModel-2021-10-06.csv)
	found: /path/to/SD-copy/LOGS/MyModel-2021-10-06.csv
f_open(/path/to/SD-copy/LOGS/MyModel-2021-10-06.csv, 32) = 0x7fff8067a748 (FIL 0x114ed8338)
f_size(0x7fff8067a748) 171057
f_close(0x7fff8067a450) (FIL:0x114ed47c0)
convertToSimuPath(): /SOUNDS/en/SYSTEM/volt1.wav -> /path/to/SD-copy/SOUNDS/en/SYSTEM/volt1.wav
findTrueFileName(/path/to/SD-copy/SOUNDS/en/SYSTEM/volt1.wav)
	found: /path/to/SD-copy/SOUNDS/en/SYSTEM/volt1.wav
f_open(/path/to/SD-copy/SOUNDS/en/SYSTEM/volt1.wav, 1) = 0x7fff8067a450 (FIL 0x114ed47c0)
eeprom write general
f_close(0x7fff8067a450) (FIL:0x114ed47c0)
convertToSimuPath(): /SOUNDS/en/thrhld.wav -> /path/to/SD-copy/SOUNDS/en/thrhld.wav
findTrueFileName(/path/to/SD-copy/SOUNDS/en/thrhld.wav)
	found: /path/to/SD-copy/SOUNDS/en/thrhld.wav
f_open(/path/to/SD-copy/SOUNDS/en/thrhld.wav, 1) = 0x7fff8067a450 (FIL 0x114ed47c0)
f_close(0x7fff8067a450) (FIL:0x114ed47c0)
convertToSimuPath(): /SOUNDS/en/SYSTEM/0000.wav -> /path/to/SD-copy/SOUNDS/en/SYSTEM/0000.wav
findTrueFileName(/path/to/SD-copy/SOUNDS/en/SYSTEM/0000.wav)
	found in map: /path/to/SD-copy/SOUNDS/en/SYSTEM/0000.wav
f_open(/path/to/SD-copy/SOUNDS/en/SYSTEM/0000.wav, 1) = 0x7fff8067a450 (FIL 0x114ed47c0)
f_close(0x7fff8067a450) (FIL:0x114ed47c0)
convertToSimuPath(): /SOUNDS/en/SYSTEM/volt1.wav -> /path/to/SD-copy/SOUNDS/en/SYSTEM/volt1.wav
findTrueFileName(/path/to/SD-copy/SOUNDS/en/SYSTEM/volt1.wav)
	found in map: /path/to/SD-copy/SOUNDS/en/SYSTEM/volt1.wav
f_open(/path/to/SD-copy/SOUNDS/en/SYSTEM/volt1.wav, 1) = 0x7fff8067a450 (FIL 0x114ed47c0)
f_close(0x7fff8067a450) (FIL:0x114ed47c0)`

`2.3.14:

StartAudioThread(11)
simuFatfsSetPaths(): simuSdDirectory: "/path/to/SD-copy"
simuFatfsSetPaths(): simuSettingsDirectory: ""
reusableBuffer: modelSel=333, moduleSetup=212, calib=110, sdManager=632, hardwareAndSettings=184, spectrumAnalyser=452, usb=512
opentxInit
f_close(0x0) (FIL:0x115fd33d8)
convertToSimuPath(): /SOUNDS/en/MyModel -> /path/to/SD-copy/SOUNDS/en/MyModel
f_opendir(/path/to/SD-copy/SOUNDS/en/MyModel) = error 2 (No such file or directory)
convertToSimuPath(): /IMAGES/trex450.bmp -> /path/to/SD-copy/IMAGES/trex450.bmp
findTrueFileName(/path/to/SD-copy/IMAGES/trex450.bmp)
	found: /path/to/SD-copy/IMAGES/trex450.bmp
f_open(/path/to/SD-copy/IMAGES/trex450.bmp, 1) = 0x7fff8067a450 (FIL 0x70000d4a6c48)
f_size(0x7fff8067a450) 1144
f_close(0x7fff8067a450) (FIL:0x70000d4a6c48)
convertToSimuPath(): /SOUNDS/en/SYSTEM -> /path/to/SD-copy/SOUNDS/en/SYSTEM
f_opendir(/path/to/SD-copy/SOUNDS/en/SYSTEM) = OK
f_closedir(0x70000d4a6db8)
opentxStart(7)
luaInit
lsScripts 0x7fb4f1d3f5f0
playFile("/SOUNDS/en/SYSTEM/0000.wav", flags=0, id=67)
playFile("/SOUNDS/en/SYSTEM/volt1.wav", flags=0, id=67)
playFile("/SOUNDS/en/thrhld.wav", flags=0, id=5)
playFile("/SOUNDS/en/SYSTEM/0000.wav", flags=0, id=16)
playFile("/SOUNDS/en/SYSTEM/volt1.wav", flags=0, id=16)
convertToSimuPath(): /LOGS -> /path/to/SD-copy/LOGS
f_opendir(/path/to/SD-copy/LOGS) = OK
f_closedir(0x70000d4a6db0)
convertToSimuPath(): /LOGS/MyModel-2021-10-06.csv -> /path/to/SD-copy/LOGS/MyModel-2021-10-06.csv
findTrueFileName(/path/to/SD-copy/LOGS/MyModel-2021-10-06.csv)
	found: /path/to/SD-copy/LOGS/MyModel-2021-10-06.csv
f_open(/path/to/SD-copy/LOGS/MyModel-2021-10-06.csv, 32) = 0x7fff8067a450 (FIL 0x115fd33d8)
f_size(0x7fff8067a450) 20631
eeprom write general`

derFliegendeHamburger, 2021-10-07:
success on macos 11.6 with toolchain installed:
- git clone --recursive  https://github.com/opentx/opentx
- git clean -f
- cmake -DSIMU_AUDIO=ON  -DPCB=X9D -DTRACE_AUDIO=ON  -DCMAKE_PREFIX_PATH=/usr/local/opt/qt5 -DPYTHON_EXECUTABLE=/usr/local/opt/python/libexec/bin/python
- make
- make package

does produce an dmg file containing a companion.app which identifies itself as 2.3.14 and the simulator has working audible audio!
And no surprise, the debug window does contain the afore mentioned sequence for each playFile() invocation.

3djc, 2021-10-11:
https://github.com/opentx/opentx/blob/2.3/radio/src/CMakeLists.txt#L37

derFliegendeHamburger, 2022-04-08:
(incomplete) list of differences between the official mac build run on github actions (which results in described error) and successful local build process posted per Kilrah's recommendation
https://www.rcgroups.com/forums/showpost.php?p=48992051&postcount=7927

Cliffweb, 2022-04-28:
Version 2.3.15 -> Sound still not working on Mac.
There is someone to work on this issue or all the dev people went to paid project ?

Cliffweb, 2022-04-28:
Companion 2.3.15 sound not working on macOS Monterey 12.3.1.
But the dev 3djc just answer on Discord "Yeah, none of us have Mac, it doesn't help"
MacOS friends -> We don't care about you ! get out of my way !

bigtwin, 2022-04-28:
This is not fair from your end. If they dont have a MAC, what do you think they should do?
None of them gets paid and all efforts are on their own. It is an open project! Why dont you assist? And do you donate to the project as a minimum contribution?
Have you ever tried a different MAC? A downgrade to test whether it is actually the OS?

derFliegendeHamburger, 2022-04-28:
As a quick fix you could follow my description to setup a current build environment for opentx (posted here in another issue) and then build 2.3.15 yourself.
Just tested with home built 2.3.15 on macos 11.6.1 and  sound is working in simulator ok (but not with the official package).

I provided some info what prerequisites were different between the official build environment and my local environment but it has gone unnoticed. My offer to work with the maintainers on this issue if they guide me is still out there. Seems they  will only accept a ready to run patch

Cliffweb, 2022-04-29:
_> This is not fair from your end. If they dont have a MAC, what do you think they should do?_
If macOS is mentioned supported then I expect the same quality level than the others platforms.
macOs version is produced by automatic process, without any quality check before to give it us.
_> None of them gets paid and all efforts are on their own. It is an open project! Why dont you assist? And do you donate to the project as a minimum contribution?_
I can't believe they are not paid, nobody work for nothing or for the glory
I have donate several time. But I stop because the issues are never solved on mac platform.
_> Have you ever tried a different MAC? A downgrade to test whether it is actually the OS?__
No, my Imac is up to date, there is no reason to destroy my configuration to do opentx testing.
I'm not beta tester of this organisation.
I'm not customer, I got opentx for free, then I have to accept the bad quality of this platform version.
The last version date to 9 months, I was hoped a miracle, but nothing happens...

3djc, 2022-04-29:
It is free software, develop by people that arent paid for it, despite what you seem to belive. If you don't like it, which you are absolutely entitled to, then you should fine another free or paid one that suits you better

Cliffweb, 2022-05-01:
Strange, sound system work well on EdgeTX, with the OpenTX wav files.
What a mess !

bigtwin, 2022-05-01:
Perfect! So you have your solution :-)