opentx/opentx #8592: Telemetry vom External MPM not processed if Internal MPM is OFF

Nihonski, 2021-07-04:
OTX 2.3.11, 2.3.12, 2.3.13
TX16S with internal and external MPM

I am using the internal MPM and additional an external MPM.
(Which I synchronize by some modification in Pascals FW via Heartbeat signal)
When the Internal MPM is OFF (or looses connection), OTX does not process the S.Port from the External module.
OTX assumes that the telemetry is lost and keeps the values from the External MPM frozen, although the external connection is stable.
The TX16S has separate Interfaces for the internal and external S.Port.
(not like the X9D+, which uses a common S.Port for both)
The two S.Ports should not block each other at any case, if one of them has trouble.

3djc, 2021-07-08:
Cannot reproduce. Working fine here as you can see (I get RSSI from receiver on second module while internal is off):

[image]

Nihonski, 2021-07-08:
Maybe I explained the wrong way.
OK, both MPMs are on and connections with RXs are stable.
I renamed the IDs of external sensors, as OTX cannot make a difference.
(Another question, can OTX automatically splitt between the internal and external telemetry with same IDs?)

see pictures:
Both RX on => everything fine.
Switch off Ext.RX => external telemetry frozen (sensor 2-6) , internal telemetry working (sensor 7-10), thats fine.
Switch off Int..RX => all telemetry frozen although external data still coming in
[image]
[image]
[image]

Nihonski, 2021-07-10:
Sorry, I accidentially closed before,
issue is not solved.

Nihonski, 2021-07-12:
To 3DJC,
I think the RSSI 88dB on your picture is frozen.

3djc, 2021-07-12:
Absolutely not

Nihonski, 2021-07-12:
Can you reproduce my pictures?

3djc, 2021-07-12:
That’s actually how it is supposed to work, one module drives telemetry on or off. What seems wrong is the choice between internal and external on those radio (rssi source)

Nihonski, 2021-07-12:
If I use the screen of your picture:
Int. RX ON, Ext. RX OFF => both RSSI (int. and ext.) indicate the same value
Int. RX OFF, Ext. RX ON => both RSSI (int. and ext.) indicate 0dB, although ext. telemetry data coming in

Int +Ext RX ON

Nihonski, 2021-07-12:
If somebody use an external Module, for redundancy purpose, and the internal module has a connection problem, then the telemetry from the external module is frozen. 
So the external telemetry cannot be used to supervise the internal connection.
For me the present functionality has no advantage to keep it like that.

Nihonski, 2021-07-12:
I am transferring the Link-Quality-Data of one connection by the telemetry of the other connection, and vise versa.
So the links supervise each other. I think thats a nice advantage, unless one telemetry connection blocks both, which happens presently.

3djc, 2021-07-12:
Clearly not how it is intended to work, hence your troubles

Nihonski, 2021-07-12:
For example, use the SBUS FL and FS bits from RX1 and transfer the status back to the TX by the other RX2.
Or the Voltages, or other data indicating the connection status.

Nihonski, 2021-07-12:
If there is trouble with one link, I will get valid log files of what happened by the other link.

Nihonski, 2021-07-15:
> Clearly not how it is intended to work, hence your troubles.

Well, if the OTX team does not want to change, I am not in a position to request or demand anything.
But its not my personal trouble.

The X9D+ had the limitation of one common S.Port Bus for both RF-modules. In that case the present functionality is understandable. With TX16 and others we got two independant interfaces for internal and external telemetry.
The former limitation become obsolete.
2.4GHz + 900MHz external module becomes more or less standard. From RF point of view, there is no problem to use both telemetry signals, even when the two RF-modules are not synchronized.

3djc, 2021-07-15:
you can use both, not an issue, just one is "master", and when master looses telem, telem lost is in place. So in your 2.4 + 900, you can decide using rssi source if 900 is your backup and you want to be on telem lost when you loose 2.4, or if 2.4 is just a short range additional telem and your telem lost is on the 900

Nihonski, 2021-07-15:
Thats ok, when the master looses telemetry, telemetry lost is in place and master telemetry data are frozen/RSSI = 0
But why should the slave telemetry data be frozen as well, although the slave telemetry works.

Nihonski, 2021-07-15:
Especially when the master link is lost and I need to rely on the slave link, I would need to know whether the slave link is also critical, dont you think so?

Nihonski, 2021-07-19:
Today I did some tests again.
When the ext. signal is lost, the external telemetry uses the RSSI from internal modul, pretending the external link is still working.
This is quite irritating and even dangerous, as I assume that the redundant link is still working, but it isnt.