opentx/opentx #8594: Ability to Name Logical Switches

jbeard6, 2021-07-05:
**Is your feature request related to a problem? Please describe.**
Maybe it's just me but, when creating models using lots of logical switches (and especially when coming back to adjust something later), it is very difficult to reconstruct my memory of what each switch does. It is also difficult to remember, on other screens, that L37 is the switch that should enable a particular mix, for example.

**Describe the solution you'd like**
Having the ability to give these logical switches a short name that appears when viewing the logical switches page and when selecting switches elsewhere in the firmware would be immensely helpful.

**Describe alternatives you've considered**
Having the ability to reorder logical switches would help remember some (e.g. "oh yeah, this group of switches controls the arming function").

elecpower, 2021-07-05:
Yaml format may address this requirement but likely only for radios that store models on sd card.
Eeprom limit for radios like x9d+ has prevented this feature and the need to be able to transfer models between radios.
Reording is available in Companion.

bigtwin, 2021-07-05:
This has been addressed at least one time [here](https://github.com/opentx/opentx/issues/7645).
Unfortunately it seems not possible to implement any kind of description to LS an SF.

jbeard6, 2021-07-09:
I hadn't seen the #7645 when I searched originally. Bummer -- it'd be really useful!

The switch to YAML sounds interesting -- is there any design documentation around that anywhere?

elecpower, 2021-07-10:
No documentation

closed 2022-04-18 by jbeard6