opentx/opentx #8595: Taranis X9D 2019 Trainer Master/Jack not working

akfpv, 2021-07-05:
**Describe the bug**
Trainer Master/Jack mode is not working in 2.3.13, tried with Internal, External and No modules selected. Radio does not detect connection, and sometimes radio reboots when scrolling to the end of the list of Trainer mode options in the Model menu. Slave/Jack works on the same radio. I tried downgrading the same radio to 2.3.7 and Trainer Master/Jack works perfectly fine. Also tested this on a Jumper T-lite flashed with 2.3.13 and Trainer Master/Jack works fine on that, so it looks like the issue is just with some radio models

**To Reproduce**
Steps to reproduce the behavior:
1. Set Trainer mode to Master/Jack
2. Connect to a Slave radio
3. Observe no channel movement in Trainer system menu

**Expected behavior**
Trainer system menu should show channel movements when connecting a Slave radio

**General information**
 - OpenTX version - 2.3.13
 - Hardware - X9D Plus 2019
 - OpenTX Companion - 2.1.23

smaller09, 2021-07-22:
same here on jumper t16. Trainer Master/Jack not working. 

OpenTX version - 2.3.13
Hardware - Jumper T16
OpenTX Companion - 2.1.23

3djc, 2021-07-22:
Working fine here on t16

smaller09, 2021-07-28:
Ok I finally make it out. I had upgrade to Type-C charger kit. the problem was caused by that. The tip of trainer jack has bad connection on the Type-C charger kit.

akfpv, 2021-07-28:
In my case the Trainer Jack is fine, tested with DVM and also it works fine as a Slave on 2.3.13, as well as a Master on older firmware (tested on 2.3.7). This is not an electrical issue, but rather a firmware issue with Taranis X9D 2019.

3djc, 2021-07-28:
Fixed in 2.3.14

closed 2021-07-28 by 3djc