opentx/opentx #8598: Your opinion: Lua scripts with touch screen?

olliw42, 2021-07-06:
hey
I'd like to ask whether this would be something you possibly would consider adding in a nearby version:

I've noted that it literally would need just few code lines to enable touch screen support and make touch available to lua widget scripts. To be clear, I am not talking about adding touch to OpenTx in general, only to read it and make it available to lua widgets.

I have this running here and I really just love it ... I bite my ass that I haven't noted this already a year ago how easy it was ...

this would add a full new layer of functionality to lua widgets, and I can't really see any cost of it

hence this question

3djc, 2021-07-06:
Not opposed to the idea personnaly, but can you tell more about what you uhave in mind ?

Reading coordinates is trivial, and would probably be enough in a full screen lua scenario, but how about widgets ? How would a widget know this swipe is for it ?

lshems, 2021-07-06:
> Not opposed to the idea personnaly, but can you tell more about what you uhave in mind ?
> 
> Reading coordinates is trivial, and would probably be enough in a full screen lua scenario, but how about widgets ? How would a widget know this swipe is for it ?

This 'focus' thing has been discussed in length and is a non issue. The issue is not what you question above, but the position opentx has taken on allowing Lua to know about button events to act on them. Position is that it will not happen.

So better be honest, instead of asking for info you'll never use.

lshems, 2021-07-06:
> hey
> I'd like to ask whether this would be something you possibly would consider adding in a nearby version:
>
> I've noted that it literally would need just few code lines to enable touch screen [...]

Did you already manage to get this running on edgeTX? I'm waiting desperately to get key and touch events in scripts and widgets.

olliw42, 2021-07-06:
>  can you tell more about what you uhave in mind ?

this is on EdgeTx, but I have exactly (= EXACTLY) the same also working on OpenTx (just the first 1:30 minutes are of relevance)
https://youtu.be/W0uaKVUnGps

I find it supercool to have buttons and selectors (and it's amazingly simple to do in the lua)

>Reading coordinates is trivial, and would probably be enough in a full screen lua scenario

yes
well, I've added few lines of code to translate into tap and wipes, but all easy too

>but how about widgets ? How would a widget know this swipe is for it ?

I don't think touch would make so much sense at this stage for smallish things, more so for "real" apps like the iNav or Yappu (or mine LOL)
so, in my mind it's only for full screen

a smallish widget could, in principle, always also figure it out by if the touch x,y are in its size
it depends a bit on how the touch data is transferred to the lua script, different approaches may have different advantages/drawbacks, but as said, I think full screen is what I' have in mind

> button event

I'm not talking about button events, but touch screen

button events are somewhat more tricky

> Did you already manage to get this running on edgeTX?

certainly not on vanilla EdgeTx
but on my flavor, see video, it's a "hack" however with respect to EdgeTx

3djc, 2021-07-06:
So getting  X,Y for full screen widget is enough for what you have in mind ?

olliw42, 2021-07-06:
> So getting X,Y touch coordiantes for full screen widget is enough for what you have in mind ?

nearly

what I do in addition is to generate TAP, WIPE_UP, WIPE_DOWN, WIPE_LEFT, WIPE_RIGHT events in the C code

in addition I currently provide these data, the original touch event, the x,y and that generated event via a lua function to the widget. As said however, this piece can be debated, could be also a function argument to the refresh() function, I just did this so since it was so easy.

that's all it took

I can push a branch to my repo if you like to inspect before voicing an opinion

> Or maybe easier to discuss a PR if simple enough ?

hmm, my experience is mixed with PRs which rather first need a general opinion ... if you say however it's not totally unrealistic I'll be happy to push a pr

3djc, 2021-07-06:
I understand about the mixed feeling with PR, my intend was just to check if that would need to cross any of our red lines, but sure, if you can get a branch up, I'll happilly look into it !

wimalopaan, 2021-07-06:
I think as said with edgetx, a full-screen (!) widget should receive all button events buts the rtn-long event. So, a script does not to have call lockKeys() UnlockKexys() as in your patch.
Additionally, it could be arraynged also to forward to swipe events.

3djc, 2021-07-06:
> I think as said with edgetx, a full-screen (!) widget should receive all button events buts the rtn-long event. So, a script does not to have call lockKeys() UnlockKexys() as in your patch.
> Additionally, it could be arraynged also to forward to swipe events.

As said previously, keys and buttons are outside the scope of the discussion, please keep it on the subject

olliw42, 2021-07-06:
>my intend was just to check if that would need to cross any of our red lines

be assured I fully understand that and hence first this issue

would be useful to understand better the red lines though

I'll do a PR with a clear sign "ONLY FOR DISCUSSION", give me a second :)

wimalopaan, 2021-07-06:
I personally would prefer an lua-api to define a touch-button as a rectangular area of the widget-zone. And then forward tap-events of this touch-button to the lua-script. Only (x,y)-events forwarding I think is too low level and place too much of a burden to the lua-side.

olliw42, 2021-07-06:
here we go: https://github.com/opentx/opentx/pull/8599

many thx for the discussion so far, much appreciated

3djc, 2021-07-08:
The PR looks interesting, made some comment on naming. Back to the main points:
- we are not hostile to the idea (on the contray)
- it is important to understand that we will likely not maintain compatibility with this in 2.4, We have plans for a more object oriented, windows based handling (each widget in its own window), but thats not for short term (actually this is the part delaying most 2.4). If thats not a showstopper for you, we can keep working on that

olliw42, 2021-07-08:
>     * we are not hostile to the idea (on the contray) in a 2.3 scope

fantastic to hear

> * it is important to understand that we will likely not maintain compatibility with this in 2.4, We have plans for a more object oriented, windows based handling (each widget in its own window), but [...]

to the contrary

personally I'm anyhow of the kind to expect that different versions work differently, potentially also fundamentally differently (I call this "progress"). I think the lats two paragraphs in this comment in the PR, https://github.com/opentx/opentx/pull/8599#issuecomment-875342581, is exactly aligned with what you are saying. So, no showstopper for me, to contrary.

olliw42, 2021-07-09:
@3djc 

you said 

> made some comment on naming

I can't find the comments, could you give me a hint where they are ? :)

3djc, 2021-07-09:
On the PR itself

[image]

olliw42, 2021-07-09:
yes, this is what I guessed, but for the heck I can't see any ... and so far/usually they always just would show up, both in the conversation and files changed...
I really feel quite stupid now, but that's what I see

[image]

anyway, I now know your comment ... I'll do this afternoon