opentx/opentx #8606: FRM302 Not working in 2.3.13

keithluneau, 2021-07-08:
I have two Radiomaster TX16S, both with OpenTX 2.3.13
I have one Jumper T16 with OpenTX 2.3.10

I have a model that is set up with a FlySky FRM302 module (module has the OpenTX compatible firmware). On the two Radiomaster transmitters, the FRM302 module isn't working. I have internal RF off, and external is selected AFHDS3. All three radios are configured the same. On the two Radiomaster radios, the module status is "Not Ready" and power source is "unknown" as if the radio is not communicating with the module. On the Jumper with older version of OpenTX it works as expected.

FYI, I did fly using the Radiomaster radios and the same FRM302 module some months ago, when they had an older version of OpenTX. I only recently updated the f/w on these two and found the FRM302 is no longer working on them.

Is there any known reason why it would have stopped working in this version of OpenTX?

3djc, 2021-07-08:
Module sync was introduced in 2.3.12. FRM302 was supposed to be compliant with that, not sure how much it really is given your comment. I don't think any of us does have such a module, the dev work has been done by @qba667. Jakub, any lights you can share with us ?

vandusinae, 2021-07-10:
I have a similar issue on a X9D+, but I can't even see the AHDS3 module in general settings and can't change the external rf to AFHDS3 (firmware file: opentx-x9d+-afhds3-lua-noheli-sqt5font-en-2.3.13-otx). It worked under my previous OpenTX 2.3.10-otx version. I switched back to that version and it worked again. Could it be deactivated for specific hardware? Tried again (adding the eu module), but no AFHDS3 module under general settings.

vandusinae, 2021-07-12:
It seems to work when I compile the current * 2.3 branch. The AFHDS3 module appears and the FRM302 module connects to OpenTX in serial mode. 

cmake -DPCB=X9D+ -DAFHDS3=yes  -DLUA=YES  -DHELI=yes  -DDEBUG=YES -DCMAKE_BUILD_TYPE=Debug ../
make firmware

version.h

#define VERSION           "2.3.13"
#define VERSION_MAJOR     2
#define VERSION_MINOR     3
#define VERSION_REVISION  13
#define VERSION_SUFFIX    ""

#define VERSION_NUMBER    ((VERSION_MAJOR << 16) | (VERSION_MINOR << 8) | VERSION_REVISION)

What is the recommended command to compile the firmware for AFHDS3, noheli, lua and sqt5font, without debug?

Thank you!

3djc, 2021-07-12:
Recommended is download a nightly tomorrow

vandusinae, 2021-07-12:
Thanks again!

3djc, 2021-07-13:
@vandusinae if you could confirm that:
1 - a nightly downloaded today does now show the FRM302 option
2 - the module works
that would be great !

keithluneau, 2021-07-13:
I'm not sure if the current nightly has anything changed for the TX16S, but it's not working for me. I installed "opentx-tx16s-lua-noheli-en-2.3.13N534.bin" on my spare radio and the FRM302 still doesn't seem to communicate with the radio. Module status is still "not ready" like before.

3djc, 2021-07-14:
@keithluneau could you try this firmware ?
[tx16s-frm302.zip](https://github.com/opentx/opentx/files/6815674/tx16s-frm302.zip)

keithluneau, 2021-07-14:
@3djc That firmware does work, the module is communicating with the radio just fine. I do not have a receiver here in front of me at the moment but I'd expect it to bind up and work. I can verify later if need be.

3djc, 2021-07-14:
Cool, thanks for your feedback. Yes please, if you have the opportunity to test end to end with a receiver and servo, that would help us greatly

keithluneau, 2021-07-15:
OK I was able to test with a receiver this morning. I used the FTr10.

Binding doesn't seem to work. If I use the menu in OpenTX to put the module into Bind mode, it immediately exits and never binds to the receiver. I put the module back into my old T16 that still has OpenTX 2.3.10 and was able to bind normally with the receiver. I then placed the FRM302 module back into the TX16S with your firmware from the post above, and was able to link with the receiver that was previously bound, and control a servo just fine. Telemetry also works.

I guess we're part way there! lol

mkania1975, 2021-07-15:
AFHDS3 option 2.3.13 does not work with my Taranis Q X7+FRM302. In the nightly version, AFHDS3 has disappeared from the external radio configuration.  I heard that version 2.3.10 worked in QX7 + FRM302, but the download of opentx-x7-2.3.10-otx is no longer possible. Can anyone help me?

3djc, 2021-07-15:
When did you try a nightly, it should be ok if you select the afshd3 option

mkania1975, 2021-07-15:
> When did you try a nightly, it should be ok if you select the afshd3 option

Thank you! I uploaded and the option appeared, the module's power supply turns on, but the status: Not ready, Power src: Unknown. So this version doesn't work with the Taranis Q X7 either :(

3djc, 2021-07-15:
I just tested with QX7, I do not have the module, but what is been output from external module bay is exactly the same as tx16s, which was found working

mkania1975, 2021-07-15:
@3djc thank you for your response. This is what it looks like https://youtu.be/LoCMq-X4sM8

keithluneau, 2021-07-15:
@mkania1975 Has it every worked in OpenTX before? You need OpenTX compatible firmware on the FRM302. I don't see a link for it on FlySky's site now, maybe contact them and ask if you haven't already done that.

mkania1975, 2021-07-15:
@keithluneau I am trying to run for the first time. I did a FRM302 software upgrade to version 0.0.2.12 made especially for OpenTX.
I made it like on this video https://youtu.be/p43xFniYZsg

qba667, 2021-07-15:
@3djc I requested once some changed to be merged to fix timing for soft UART on Q7, Betranad never merged them.
@mkania1975 @keithluneau guys are you sure you are using 115200 bps firmware?

qba667, 2021-07-15:
@3djc can you take a look at https://github.com/opentx/opentx/pull/8019

3djc, 2021-07-15:
Isn’t there unanswered question on this one ?

mkania1975, 2021-07-15:
> @mkania1975 @keithluneau guys are you sure you are using 115200 bps firmware?
@qba667 Yes I'm using 115200 version 0.0.2.12 firmware in my FRM302 module.

qba667, 2021-07-15:
No, all was answered and clarified. Keeping of previous priority is optional action, up to you what you want to do, we can eventually always use higher priority.

3djc, 2021-07-15:
Will have a look

qba667, 2021-07-15:
@3djc  thanks!

vandusinae, 2021-07-15:
the latest nightly build seems to work on the X9D+, X-Lite and X-Lite S radios.

3djc, 2021-07-16:
@qba667 looked briefly at #8019, looks to me like you are switching between oldPriority and 0, but looks also like TELEMETRY_EXTI_IRQn priority is actually set at 0 both on Taranis and Horus inn current code

3djc, 2021-07-16:
@qba667 could you have a look  at https://github.com/opentx/opentx/tree/3djc/F2-softserial and test it if you have the required equipement ?

mkania1975, 2021-08-08:
I don't know what was changed, but on version 2.3.14 my Taranis Q X7 works fine with  FRM302. I thing you can close this issue.

3djc, 2021-08-08:
Indeed we fixed it

closed 2021-08-08 by 3djc

khmewes, 2024-11-13:
Sorry guys the problem of not binding is back after all this time. It's back in OTX version 2.3.15. I can't bind to a FTr10 with the FRM302 module FW V. 2.12 115200.  Please anyone who is still receiving this and has the time take a look at it again.

khmewes, 2024-11-13:
> @3djc That firmware does work, the module is communicating with the radio just fine. I do not have a receiver here in front of me at the moment but I'd expect it to bind up and work. I can verify later if need be.

Hi, how do I flash this binary to the FRM302?