opentx/opentx #8608: Horus X12S - 2.3.13 servo movement jittery/delayed

dl3it, 2021-07-08:
Radio used: Horus X12S, internal ACCESS module, ACCST mode; all components flashed with latest firmware

Using opentx 2.3.11, everything works fine.

Using opentx 2.3.13, servo movement is jittery and sometimes up to 1s delayed. I tried several receivers, X8R, RX8R, XM+, ... same problem. Servo movement is a bit smoother when using D8 mode, D16 mode is unusable.
Interesting sidenote: I cleaned the whole radio; new formatted sdcard, standard setup, automatic generated model. Everything works fine. After adding the next model (with built in wizzard), the new model shows jitter, the first model remains ok. Every additional model shows the problems; no difference between models new created or imported by the companion.

3djc, 2021-07-08:
Please try nightly N530 or later

kilrah, 2021-07-08:
Duplicate of #8575

closed 2021-07-08 by kilrah

dl3it, 2021-07-08:
seems to be fixed :-)

Thanks for the super fast reply...