opentx/opentx #8610: MPM protocol numbering missmatch?

olliw42, 2021-07-09:
concerning the MPM, the numbering of the protocols listed by MPM in the Multi.txt file (https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Multi.txt) and by OpenTx in the modules_constants.h file (https://github.com/opentx/opentx/blob/2.3/radio/src/pulses/modules_constants.h#L101-L194) do not appear to be consistent. 

See e.g. the entries for 83 and 86.

I have no idea what the implication of this is or if it has any, but I thought I should mention...

3djc, 2021-07-09:
Early in the dev on multi, it was deemed easier for the user to regroup in Otx UI all the FrSky protocols under one entry, so it looks like a XJT in a way. The issue was that for MPM, V, D , X versions where different protocol numbers. So a mechanism was created to have different numbers. This is years old, and all user setup actually store otx protocol numbers, not MPM? Pas cal and I are frustrated by this almost daily, and we will change that, but it requires an eeprom conversion so will happen in 2.4

closed 2021-07-09 by 3djc

olliw42, 2021-07-09:
many thx for the explanation

I failed to see that there is an additional offset by 1 between the tables, which I just now have realized

(i.e., in MPM 86 means CONFIG, which in OTX is listed as 83, but convertMultiProtocolToOtx() is subtracting 2, converting 86 into 84 and not 83 ... but an additional -1 is subtracted when used)

sorry for this