opentx/opentx #8617: channel 3 ghost mixing on channel 2 after mix removal

nagordon, 2021-07-13:
I have been using my frsky x10e for about 1.5 years now with no issue. A few months ago I added a elevator mix to the throttle, such that when I add throttle, the elevator pitches down with a delay. A few days ago, I removed this mixing, and now, there is a undefined/ghost mix that is occurring only on channel 2. The elevator still moves the channel but now also the throttle mixes with the elevator as well, even when there is no mix defined. The worst part of this is that every single other model on my radio now have this ghost mix with the throttle adding to the elevator channel 2! I've tried recreating new models and it still has this issue.

here is the setup before I removed the mix
[image]

[image]

here is the radio with a new model recreating the undesirable mixing that is occurring without there being any mixing in opentx. The video shows the inputs page with just 3 inputs
[image]

For the Mixes I created 4 new mixes (with channel 4 repeating the use of input 2. CH1 and CH3 work fine, but CH2 has odd mixing from the elevator stick and throttle stick.

[image]

I have recreated this issue on new models. I've tried updating to 2.3.13 but no change. 

**General information**
 - OpenTX version open tx 2.3.10 
 - horus x10 express

Thanks

3djc, 2021-07-13:
You do not specifiy at all what you do where. Are you editing your model with companion ? If yes then can you try doing the same on the radio and report ? Also please provide a step by step so we have a change to reproduce

nagordon, 2021-07-14:
> You do not specifiy at all what you do where. Are you editing your model with companion ? If yes then can you try doing the same on the radio and report ? Also please provide a step by step so we have a change to reproduce

I have created new models with the wizard on the radio, and new models in companion. I am not editing the models at all, just creating new ones, and showing the elevator and throttle channel mixing when there is no mixing. I have removed all models from my radio. 

After performing the stick calibration on the fresh install, you can see the throttle and elevator mixing (see animation)
[image]

I did try the companion simulator on the model that this issue first occurred, and the ghost mixing did not happen and the model behaved as expected. Is it possible it is some sort of hardware issue on my X10e?
the companion model is here
[zoho dart.odt](https://github.com/opentx/opentx/files/6813390/zoho.dart.odt)

frankly i don't know how to replicate it because I cannot get it to go away in the first place. The best I can describe how to replicate it is to create this model on a frsky X10e, and remove the throttle mixing(which had slow ups) on the elevons, that is when this issue appears. 

this discussion feels relevant to this problem, but I am baffled on how to test it since I only have 1 "infected" radio
https://www.openrcforums.com/forum/viewtopic.php?t=8648

nagordon, 2021-07-14:
So I installed the ethos bootloader and firmware on the radio and the same exact thing is happening!!! I am convinced it is a hardware error and not an opentx bug.

3djc, 2021-07-14:
Ok, closing this then,Best of luck to get your radio repaired

closed 2021-07-14 by 3djc

nagordon, 2021-07-17:
well, I don't think anyone will ever believer me, but when I unplugged and replugged both gimbals, everything works fine now. If anyone can explain how this happened I will buy you a beer.

kilrah, 2021-07-17:
Bad contacts happen... probably not more than that.
Would really wiggle the wires and connector a bit while moving the sticks and check it's reliable, last thing you want is that a wire/solder point is broken and it reappears in flight after you move the sticks a bit.

nagordon, 2021-07-23:
I am just curious how a hall effect sensor could "mix" the channels by missing a contact. Must be something on how they come together on the circuit board?

kilrah, 2021-07-23:
When you have an ADC with multiple muxed inputs (like here) if one input is not connected it will read about the same as what the previous one that was read.

nagordon, 2021-07-25:
> When you have an ADC with multiple muxed inputs (like here) if one input is not connected it will read about the same as the previous one that was read.

ahh, ok. that's make a little more sense. hopefully this will serve other people with a similar issue. thanks