opentx/opentx #8618: T-Lite Trainer external module does not work

0815Creeper, 2021-07-13:
**Describe the bug**
T-Lite with sbus receiver hooked up to the external module bay does not receive a trainer signal. RX can be turned on and off selecting or deselecting the master/sbus setting as it should but trainer mix lines stay deactivated and trainer mode does not work. Same for ppm receivers and ppm setting

**To Reproduce**
Steps to reproduce the behavior:
1. Plug sbus receiver into heartbeat of external bay
2. Set trainer mode in active model to master/sbus, the RX turns on (green light if slave radio is transmitting.
3. Check otx trainer page for moving numbers, they don't.

**Expected behavior**
Moving numbers on the trainer page if RX bound to sending slave transmitter and moving it's sticks.

**General information**
 - OpenTX version 2.3.13
 - Hardware T-lite

0815Creeper, 2021-07-13:
Possibly a HW defect but someone else would have to check that, I only have 1 t lite

3djc, 2021-07-13:
I just had a look at tlite schematics, and by the look of it, there is a reversed diod and missing vcc connection for this pin to actually work... Will contact Jumper in case my drawings I wrong ....

0815Creeper, 2021-07-13:
Damn, thanks. Any chance this can be HW moded to work? At least in ppm mode?

3djc, 2021-07-13:
ppm or sbus is the same issue, the pin goes trough an hardware inverter, and that circuit is all wrong. Sure it can be hardware modded, it requires D402 to be reversed, and a 3.3v brought between R430 and R429? Where are those ? Your guess is as good as mine !

0815Creeper, 2021-07-13:
I see, thanks. I am looking forward to Jumpers answer on this.

brainbubblersbest, 2021-07-13:
Using a RXSR connected to it's uninverted SBUS pin wont make the Trick? As it is the way on the most F4 Flight Controllers Today.

3djc, 2021-07-13:
No, it’s not that the inverter doesn’t invert, it let’s no signal through

brainbubblersbest, 2021-07-13:
Because of the reverted Diode it blocks + Potential?
But a low Signal can let + Potential through the Diode.
Dont know the Schematic and have no t-lite. But that comes into my mind.

brainbubblersbest, 2021-07-13:
Ive read Jumper dont Support frsky r9m lite for that Radio.
Maybe thats why they done this to make shure the weak supply Trace dont get overloaded. The tbs is only supported with 50mw tx power😕
Instead of 7 pin connection like on the x lite, that contain additional two Pins for two wire uart, 
here is only: sbus, blocked heartbeat, vbat, gnd, and s.port

Weird Design....

3djc, 2021-07-15:
Ok, I have tracked down the issues, yes, there are two:
1) a VCC pullup is missing. It can be fixed with a hardware patch like this
[image]

2) the input pin on the external module 'thing' is not connected at all
[image]
This one is a mess to fix, so i did not do it

3djc, 2021-07-15:
It needs a hardware fix for a missing VCC pull up

[image]

closed 2021-07-15 by 3djc

kilrah, 2021-07-15:
> Ive read Jumper dont Support frsky r9m lite for that Radio.
> Maybe thats why they done this to make shure the weak supply Trace dont get overloaded. 

No it's because the R9M lite talks over 2 pins of a fast UART and Jumper haven't brought one there.

0815Creeper, 2021-07-15:
just to clarify before i start soldering: the only thing that needs to be done is soldering this wire as shown in the picture?
The necessary pin definitons are done in otx?
There is no reversed diod?

3djc, 2021-07-15:
No, diod was reversed in the schematic, but not in boards. All is needed is this wire

0815Creeper, 2021-07-15:
Works great with the mod. Thank you very much

brainbubblersbest, 2021-07-15:
> > Ive read Jumper dont Support frsky r9m lite for that Radio.
> > Maybe thats why they done this to make shure the weak supply Trace dont get overloaded.
> 
> No it's because the R9M lite talks over 2 pins of a fast UART and Jumper haven't brought one there.

Yes, I know. But why they planned to use a xlite Bay then?
Only for multi and crsf nano? 
Heartbeat Pin is not really need here. Seems more they finally decide to change wiring before the release, so no one use a r9mlite at 500mW. Jumper advertised Crsf Support with more Output at the Release too. And corrected this a short time later.
Isnt the pxxrx pin bridged with the HB pin in that external pcb?
[image]
[image]

0815Creeper, 2021-07-15:
> Heartbeat Pin is not really need here.

yes it is needed, as the SBUS or PPM RX for trainer operation is connected here. 

Please open a new issue if you think there is a problem with the r9m or any other module. This has nothing to do with modulebay-tranier functionalty

brainbubblersbest, 2021-07-15:
The pxxrx is used for Trainer input on other Taranis Radios. its just a bit investigating what Happening here, leading you need to mod a T-Lite for SBUS input 😉 I would not say it has nothing to Do with that. But at least youre Happy with it.

0815Creeper, 2021-07-16:
Ok, enlighten me please: which radios? On all radios with a JR bay the RX is connected to heartbeat.
Aren't there only the xlite and x9lite remaining? Do you connect an sbus RX to their pxxrx instead of heartbeat?

brainbubblersbest, 2021-07-16:
Colour X10 express for example. But I dont have one. The Fullsize JR Module Bay ditch the sbus out and HB pin for Serial tx and rx.
And for a BW Taranis based example.

derFrickler, 2022-07-12:
I use the Module Bay of my T-Lite for ELRS, but want to use the wireless trainer via the internal Multi-Module.
Is there a way to get this working? 

If i get it right, the normal FrSkyRX-Multi mode on the 4in1 is notworking, as internal and external module share the same Telemetry line on most Transmitters (T-Lite seems to have the same schematics as the X7?)

The Multimodule can be run as RX with CPPM output on the STM32 USART1.TX pin (BOOT0 programming TX pin)

3djc, 2022-07-12:
No reason for it not to work with multi trainer

3djc, 2022-07-12:
indeed

derFrickler, 2022-07-12:
Great, works now, X7 as Student, T-Lite with FrSkyRX-Multi to the Student and ELRS to the Quad.