opentx/opentx #8626: sending smartport telemetry to s.port pin

RomanLut, 2021-07-22:
I am trying to send smartport telementry to the s.port pin on Jumper T-lite from ESP32.

Internal RF is off, External RF is set to SBUS.

In documentation I was able to find, smartport protocol is described as pooling protocol: Receiver on model is pooling sensors by sending packet {0x7E device_id:8}, while sensor/FC replies with a smartport packet {0x10 id:16 value:32 crc:8}.

Do I understand correctly: the pooling is performed between receiver on the model and FC only? S.port pin on Jumpter T-lite should be like "sniffing" this communication? I am aking because I do not see any bytes coming out of S.Port pin on Jumper T-lite. It looks like RX only pin.

I am able to send packets from ESP32 to S.port pin of Jumpter T-lite every 11ms:
this->serial->write( 0x7e );
this->serial->write( 0x98 );   
this->serial->write( 0x10 );   
this->serial->write( 0x01 );   
this->serial->write( 0xf1 );   
this->serial->write( 0x64 );   
this->serial->write( 0xcd );   
this->serial->write( 0x00 );   
this->serial->write( 0x5e );   
this->serial->write( 0x6c );   

and discover RSSI sensor. 

Do I understand correcly: S.port pin is RX-only pin, OpenTX can acccept multiple telemetry protocols on S.PORT pin, it detects telemetry protocol by analysing incoming packets?

kilrah, 2021-07-22:
> Do I understand correcly: S.port pin is RX-only pin

The radio can send things to the receiver / devices for configuration, but yes for telemetry the radio just listens. 

> OpenTX can acccept multiple telemetry protocols on S.PORT pin, it detects telemetry protocol by analysing incoming packets?

No, it selects what to use based on module type/protocol.

closed 2021-07-22 by kilrah

troisiemetype, 2021-08-11:
@RomanLut , did you manage to get the telemetry running from the esp32 ?

I'm facing the same issue, being implementing an esp32-based module to fly Parrot Minidrones. For what I understand trying to read through the openTX code, Multiprotocol and other sources on the net, both the request from receiver to sensors and the answer from sensor to receiver should be echoed by the external module on the S.port pin ?

I can't have any reaction on my Taranis X7 when trying to make sensor-like frames. I've tried both with sbus and PPM out. With PPM I can choose the telemetry type, but that doesn't change anything.

Maybe there is a need to implement a heartbeat or handshaking between module and transmitter at startup ? Is this documented somewhere ?

RomanLut, 2021-08-11:
Yes I managed to send RSSI, A1 and A2. Yes you have to send both request and response.
See SmartPort::loop(): here:  [https://github.com/RomanLut/hx_espnow_rc/blob/main/examples/esp32_devkit_v1_tx/src/smartport.cpp](https://github.com/RomanLut/hx_espnow_rc/blob/main/examples/esp32_devkit_v1_tx/src/smartport.cpp)

You can find more sensor examples here:
[https://github.com/multiwii/multiwii-firmware/blob/upstream_shared/Telemetry.cpp](https://github.com/multiwii/multiwii-firmware/blob/upstream_shared/Telemetry.cpp)

RomanLut, 2021-08-11:
There is no need to handshake. You can discover sensors onse you send the data periodically. Note that SPORT is inverted uart:
this->serial->begin(57600, SERIAL_8N1, -1, SPORT_PIN, true );

troisiemetype, 2021-08-12:
Hi Roman.
Thank you for the advices, and for the links to your code and MultiWii. I already spent some time reading different parts of your code, that helped me to understand how it's intended to work.
I don't know how many times I've checked the uart settings, the pinout, and various other things.
I can't have it working, I'm wondering if there is simply not a compilation option on my transmitter that disables the s.port pin.

Shall you wish to have a look at my code, you can find it here : https://github.com/troisiemetype/ParrotESP
It's rather messy for now, some cleaning needed.

RomanLut, 2021-08-12:
Just send:

this->serial->write( 0x7e );
this->serial->write( 0x98 );   
this->serial->write( 0x10 );   
this->serial->write( 0x01 );   
this->serial->write( 0xf1 );   
this->serial->write( 0x64 );   
this->serial->write( 0xcd );   
this->serial->write( 0x00 );   
this->serial->write( 0x5e );   
this->serial->write( 0x6c );  

every 11ms and you will be able to discover RSSI sensor.

troisiemetype, 2021-08-12:
Ok, I've got it working.

It didn't because I didn't include the header byte 0x10 when computing the CRC. In openTX the crc is computed letting the first byte apart, I must have missed something elsewhere. In MultiProtocol the first byte is also not used, but for other reasons. This is quite confusing. :)

I'm very grateful to you !