opentx/opentx #8627: Companion 2.3.13 and Windows 10: Companion gradually slows down with extended usage.

Miami-Mike, 2021-07-23:
I've found that when using Companion 2.3.13 over long periods to edit and simulate model setups, the program gradually slows down, with ever-growng lags between mouse clicks and program response, until it becomes virtually unusable. At that point I can save my working otx file, close Companion, open it again, reload the otx file, and everything will be normal until the lag slowly builds up again.

Using Task Manager, I've found that right after starting Companion and loading an otx file, memory usage is about 25 MB and power usage is "Low".

[image]

As the lag between mouse clicks and program response grows during use, I've found that memory usage also grows, and power usage progresses from "Low" at program start to "Very high" when lag is extreme.

[image]
  
I've seen memory usage climb to over 500 MB and it would presumably have climbed higher if I had let Companion keep running.

I don't know much about how Windows 10 works, but my impression is that some fault in Companion is causing it to consume an ever-increasing amount of computer resources. I don't recall ever encountering another program that behaved like this.

Miami-Mike, 2021-07-23:
Here's an interesting experiment to try:

1. In Windows 10 Task Manager, select "View" > "Group by type".
2. Start Companion. It should show at the top of the Task Manager list.
3. Click "New", start a new otx file, and add one model.
4. Task Manager should show about 25 MB memory and "Very low" power usage for Companion.
5. Open the model for editing. Task Manager will show an increase in memory of about 25 MB.
6. Close the editing window. Memory does not decrease!
7. Open the same model for editing again. Task Manager will again show an increase in memory of about 25 MB.
8. Repeat opening and closing the same model (or any other model) for editing and you'll see an increase in memory of about 25 MB each cycle. After a few dozen cycles I got it to go past a gigabyte.

[image]

bigtwin, 2021-07-23:
I checked it and can confirm the behavior on MAC OSX Mojave and Companion 2.3.13. It is just the same, but the deal happens with more memory. Each time I open a model for editing, memory consumption increases by about 70MB. However, the memory is released when I close the OTX file. 
While it does not harm my usage of Companion, it may is interesting for a developer. Is this a memory leak in the code or does it behave as designed/expected.

kilrah, 2021-07-23:
Interesting, looks like something isn't freed on model window close.

It's been there forever, even there in 1.52 from 2014.

elecpower, 2021-07-23:
Its been that way for so long it has now become a 'feature' lol.
I did some memory diags a while back trying to trace another issue and noticed consumption does grow as identified but the tool reported a relatively small memory leak once Companion is closed.
The fact that memory is eventually released is likely to point towards Qt's thread and memory management. From experience trying to force freeing of Qt resources before it wants leads to Companion crashing. If this is the case then a code refactor might be the only solution. Anyway leave the issue open for now until time permits to conduct preliminary investigation to determine the effort to fix.

PR #8851 (2022-01-30): Companion recover memory on model edit dialog close
Partially fixes #8627
Forces memory recovery on dialog close.
closed 2022-01-30 by 3djc