opentx/opentx #8631: Bluetooth PIN Code showing in Hardware just default Value

hpoorthuis, 2021-07-24:
others already reports similar Issues before:
https://github.com/opentx/opentx/issues/6440
https://github.com/opentx/opentx/issues/7381
https://github.com/opentx/opentx/issues/7381

I did build the FrSky Bluetooth Module in to MY TX16s
[image]
and yes I got the Telematik Data, until the Module close the Connection because of Pairing failure.

The Pin code showing is just Hardcode: but my Modem seems not to like the default Pin Codes like  0000 or 1234 or 1111 or 000000. This may functions with the HC06 /HC08. but not with the FrSky Bluetooth Module for Horus , witch fits and function in the TX16S
[image]

I spy a little bit on the Port and in opentx code:
Commands send to the Modem:
[06][06]fž†[18]fff`[06][1E]æ€˜€AT+NAMEtelemetry
AT+TXPW0
AT+ROLE0
ANSWERS:
Peripheral:4C24983834F6 and some OK

to get the Correct Pin code to Show it have to be Requested, But in the OpenTX Code there is no AT+ PSWD? command!
further Request would be that the Pin Code can be Set by the Hardware Menu (AT+PSWD=Param1)

Set/Inquire passkey
AT+ PSWD?
AT+PSWD=Param1

 - All OpenTX version (e.g. 2.3.xx)
 - Hardware RADIOMASTER TX16S with Horus X10/X10S Bluetooth Modul

3djc, 2021-07-24:
The firmware in FrSky module is proprietary and afaik does not allow pw change

chemata, 2021-07-24:
And what pin works?
In mine does not work the "000000" -
It is impossible to connect it to the phone (android)

hpoorthuis, 2021-07-31:
what about a new feature , 
at least try to read the pin and show on success the correct code instead of the "000000"
AT+PSWD? should do the work on the most Modems

3djc, 2021-07-31:
As I said, password on FrSky BT firmware is loocked to 000000, and there is no command to read it