opentx/opentx #8633: New Inputs are not working

djabmal, 2021-07-24:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
when you want to create a new input ( for dual rates for exemple) the new input is not taken in account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '.inputs..'
2. Click on '..ail; insert after ..'
3. Scroll down to '.weight or expo ...'
4. the values are not taken

**Expected behavior**
to have the possibility to have several inputs for one command (aileron or elevator.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version  2.3.11; 2.3.12 ;2.3.13)
 - Hardware (TX16S)

3djc, 2021-07-24:
Please spend some time learning how mixer works in a multi line scenarios. You can get help for the reading OpenTX university, or using community sites like rcgroups or openrcforum

closed 2021-07-24 by 3djc

djabmal, 2021-07-24:
> 
> 
> Please spend some time learning how mixer works in a multi line scenarios. You can get help for the reading OpenTX university, or using community sites like rcgroups or openrcforum

Hello i have used opened for the last year and tried to learn as much as I can . In this case it seems really that it is a real bug …..
 1) I can had inputs in companion and it  works in companion but as soon as I transfer it to the radio it bugs again

2 I can add input at the end of the list and it works but I can’t add an input under a line like aileron for exemple …..

Maybe you could try to help?

Thanks

kilrah, 2021-07-24:
In inputs only the first line that has a true condition is active, always been like this. You're probably confusing with the mixer.