opentx/opentx #8634: GPS + BT

chemata, 2021-07-24:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

chemata, 2021-07-24:
For the Radiomaster TX16S OpenTX Companion it does not stop configuring with GPS and Bluetooth simultaneously.

3djc, 2021-07-24:
It does here

3djc, 2021-07-24:
https://user-images.githubusercontent.com/5167938/126875446-71f8d58e-b29c-4691-9fc2-db8a5558d813.MOV

3djc, 2021-07-24:
[IMG_0955.zip](https://github.com/opentx/opentx/files/6872777/IMG_0955.zip)

closed 2021-07-24 by 3djc

chemata, 2021-07-24:
Very good.
Now activate both at the same time

3djc, 2021-07-24:
how exactly ? clicking on one desactivate the other

chemata, 2021-07-24:
That is what happens. But the Radiomaster TX16S has 2 UART ports. BT + GPS can be put internally. But if the firmware doesn't support it ......

3djc, 2021-07-24:
> For the Radiomaster TX16S OpenTX Companion it does not stop configuring with GPS and Bluetooth simultaneously.
 Then why do you complain that Companion does NOT stop you configuring both when it does. 

BT only works with Horus BT module that required a specific port, so it need someone to develop further the internalgps option to be able to choose the UART port, but it starts to get messy

chemata, 2021-07-24:
That's not true at all. There is a video from a Russian who puts GPS and BT HC-05.
From what you say, if I have BT FrSky for Horus in OpenTx companion I only have to add the internal GPS, and the BT detects it alone?

3djc, 2021-07-24:
BT option is for FrSky BT module ONLY. If you add a BT HC05 then it acts as a transparent UART and does not require an option

chemata, 2021-07-24:
I don't think I can explain myself well.
I do have the BT on the corresponding site. The FrSykyBT module.
Now I want to put the internal GPS. From what I understand I have to configure the OpenTX with the internal GPS option. The BT is transparent and detects it by itself?

3djc, 2021-07-24:
No, you cannot do that, it would need modification to both OpenTX and Companion. GPS and HC-05 is however possible

chemata, 2021-07-24:
I don't think we understand each other.
That's the problem.
OpenTX Companion does not allow to generate the Firmware of the radiomaster TX16S with internal GPS and internal BT module (FrSkyBT module).
I think OpenTX companion has to leave the 2 options simultaneously

3djc, 2021-07-24:
You CANNOT use Horus BT module and GPS at the same time without important otx and companion change. You can make a request for that change

chemata, 2021-07-24:
Now we have understood each other. That is precisely what I am trying to do. Maybe I got the wrong place, but I don't know where to expose it.
But I sincerely thank you for the kind response and help from him.
I hope someone responsible has time to read this and can implement both options in the future

AlksSAV, 2021-07-24:
> You CANNOT use Horus BT module and GPS at the same time without important otx and companion change. You can make a request for that change
I use BT and GPS, in TX16S.  In the firmware, I specified the GPS version, and BT via UART to the lower connector(mirror mode).  I recorded a video 
https://youtu.be/0UulXKpc6Kg

3djc, 2021-07-24:
That’s not an horus bt module like he has

chemata, 2021-07-24:
Thank you Alexander for your intervention. I was talking about your video precisely. What happens is that your language (I think Russian) I do not understand, and there is no possibility of simultaneous translation.
I thought you made both BT + GPS work.
I hope that in future versions of OpenTX it can be used together.
No, it is not the same module, but it uses both uarts. Another thing is that you do not work both at the same time

chemata, 2021-07-24:
Alexander,
I have tried joining your video with this one:
https://www.youtube.com/watch?v=a3wXHn0cokg
But this one doesn't work

3djc, 2021-07-24:
BT and gps at the same is not an issues, horus bt and gps is

chemata, 2021-07-24:
How ?, if openTX Companion does not stop?

chemata, 2021-07-24:
you configure GPS or configure BT. Both of us do not

3djc, 2021-07-24:
As I said earlier you only need the BT option for horus BT, not for hc 05, so if you use hc05 and gps, you only need gps option

chemata, 2021-07-24:
You will forgive my ignorance.
I have installed the BT FrSkyBT for Horus on the site that the Radiomaster TX16S has for it.
I also have a GPS (BN-800) installed in one of the two UARTs. According to OpenTX Companion the two cannot be used together. You can't both get dizzy.
That alone is my problem. Comrade Alexander did it with another BT module, but I don't understand how. Not my problem.
But if Alexander makes it work, I'll ask him for advice. Or subtitles your video

rotorman, 2021-07-24:
The protocol that talks to FrSky BT module is different than the protocol that talks with external HC-05.

For FrSky BT you need to activate in OpenTX the special build option "bluetooth", for HC-05 no such option is necessary - you just connect and use it via std. serial protocol (USART).

OpenTX code is hardwired to use for internalgps and for FrSky BT USART6 (also shared with AUX2 port). This is the reason why you currently can only choose internalgps or bluetooth, but not both at the same time.

What you want is to connect FrSky BT (internally to USART6 / AUX2) and connect GPS to USART3/AUX1. But the current code does not support GPS yet on USART3/AUX1.

I hope this clarifies the issue a bit.

chemata, 2021-07-24:
Spectacular.
Very well explained.
It has become very clear. I hope that soon it can be done.
Thank you very much

chemata, 2021-07-24:
I think the problem has been clarified.
A shame, but cleared up. I guess this question can be ended. What I do not know if I have to close it because I am the one who has started it or if an administrator has to close it.

rotorman, 2021-07-24:
Glad to hear you understood the problem. The issue has already been closed, so no further action is necessary from your side!