opentx/opentx #8637: Companion quits (crashes?) when opening an otx file

MikeNH, 2021-07-25:
I've just installed version 2.3.14 of OpenTX Companion but this was also occurring in the previous version. 

Occasionally I get the message "companion update check failed, new version information not found" when first opening Companion. When attempting to recreate this message, things seem to function normally either reporting that Companion is up to date or there is a new version available. I don't know if this is part of the real problem of Companion crashing.

Steps to reproduce crash:
1 - Open Companion and create new otx file.
2 - Save otx file. 
3 - Close otx file
4 - Open otx file just created.
NOTE: In creating these steps I discovered that document1.otx causes Companion to crash but document2.otx appears to function normally. Document1.otx was created with previous versions of Companion.
[otx_documents.zip](https://github.com/opentx/opentx/files/6874384/otx_documents.zip)

The way I've been working around this is by editing the document quickly when it opens. When it opens, the hour glass spins for a few seconds and then the crash occurs. If I start editing before the hour glass stops spinning, I can work with the models in the otx.

Thanks
Mike

elecpower, 2021-07-25:
@MikeNH can you please clarify 'crash'. Is it the error message or the Companion application terminates and needs to be restarted.
The error message you are getting usually means Companion cannot access the OpenTx web server to check for updates.
Also what OS and version are you running?

elecpower, 2021-07-25:
@MikeNH also your 'NOTE' above does not make sense re 'previous versions' as your steps don't use an old version to reproduce. the issue

MikeNH, 2021-07-25:
First, thanks for the prompt response.

"can you please clarify 'crash'"
The Companion process when viewing Windows Task Manager terminates after the hour glass spins for about 5 seconds. Is that a crash????

"The error message you are getting usually means Companion cannot access the OpenTx web server to check for updates."
Fine. I just wanted to give complete information in the event there is a relationship. The error apparently is independent of the shutting down of Companion.

"what OS and version are you running?"
Windows 10 Home. Version 21H1.   I install all the updates.

kilrah, 2021-07-26:
Can't reproduce with those files or otherwise.

MikeNH, 2021-07-26:
So this is an anomaly in my environment. I can do a workaround. Thank you for looking at it.

MikeNH, 2022-01-29:
I'm reopening this issue because the workaround referenced above no longer works.  Attempting to quickly edit the otx file doesn't open it for editing. Companion is unusable at this time.

I've tried to understand the debug logs but they are beyond me. Here are the first lines that reference "error":

_[D] StorageFormat::setError():77 - [bin] error: Invalid binary EEPROM file C:/Users/MRowe/Google Drive/Quadcopters/OpenTX 2.3/SD Card/MODELS/TEST.otx
[D] EepeFormat::load():38 - [eepe] No EEPE header
[D] StorageFormat::setError():77 - [hex] error: Invalid EEPROM File C:/Users/MRowe/Google Drive/Quadcopters/OpenTX 2.3/SD Card/MODELS/TEST.otx_

Does this debug file address the problem? 
- I've reinstalled Companion after cleaning out previous installation. 
- After finding a similar bug solution I removed any reference to previous versions in the Windows Registry.
- Using all the Companion "Read/Write"  options didn't solve the problem.
- Other attempts also.
- The Frsky XLite-Pro is working fine in spite of writing and reading to it.

This is a URL to a video that demonstrates the problem. In the video a new otx file is created, named and closed. Opening it for the 2nd time and all subsequent times demonstrates the "crash".  Is there another debugging option?

[CompanionDebug_22-01-28_22-34-58.log](https://github.com/opentx/opentx/files/7964401/CompanionDebug_22-01-28_22-34-58.log)
[Crash Video](https://rumble.com/vtl0i8-opentx-crashes-opening-otx-file.html)

bigtwin, 2022-01-29:
I can **not** reproduce it on my Mac. However, I observed than I need to add "otx" as a file extension to the filename. If I dont do it (as you in your video) the save does simply not work. The file is not being saved. 
Would your test case fail as well if you add otx to the file name?

MikeNH, 2022-01-29:
The Windows application (OpenTX Companion) automatically adds the appropriate extension.  
[image]