opentx/opentx #8639: Companion 2.3.14: Can't edit calibration settings.

Miami-Mike, 2021-07-28:
In Companion 2.2.4 and earlier versions you could edit the Negative span, Mid value, and Positive span numbers in the Calibration tab of the Radio settings window. You're no longer able to do that.

elecpower, 2021-07-28:
This was a deliberate decision as the values are set during the hardware calibration process on the radio.
On the surface manually overriding would appear contrary to the reason why calibration is performed.
What is the use case for tweaking in Companion?

MRC3742, 2021-07-28:
I've found the mid value useful for realigning stick centers.
After a new calibration the stick centers will sometimes vary. After moving the stick and slowly bringing it back to center from both directions it will vary a few points from center. 
Having the mid value adjustable I was able to move the TX center point to a true stick center.

Not sure if its needed but it was nice to be able to tweak these calibration values.

Rich

elecpower, 2021-07-28:
@MRC3742 thanks for the use case. 
IMHO Companion should compliment the radio functions and not be able to perform such a fundamental function that the radio cannot.
Needs more discussion.

mpjf01, 2021-07-29:
I use the feature to adjust the midpoint of a 6 position switch in X12S and X10S. I use it to vary the endpoints for the sticks in my X-Lite as the calibration feature does not do that perfectly. I use it to adjust the minimum throttle value if I can't get it right using the normal calibration routine.

If the problem just is that the facility is available on Companion and not on the transmitter I observe that it might just be the exception that makes the rule. or, if someone thinks it is that important to have the same exact features available in Companion and on the Tx, that, in this case, would be a notion driven more by idealism than anything else. I have yet to hear of any practical concerns or problems caused by the previous arrangement that seems to have been operative since at least 2013.

Without wishing to be negative or provocative, the appropriate time to have the discussions about removing the feature would have been before it was done, not after. Now we are in a situation where we have to beg to get it back and justify why the developer's decision to remove it was inappropriate.

There always should be a way to track changes from initiation in the Github issues list to completion. That's why the system uses Github. 

The idea that developers or anyone else should be making functional changes to the firmware without wide discussion, or consideration of users needs, or documenting on Github, goes against the grain - for me anyway.

3djc, 2021-07-29:
Everything I read make me think calibration on the radio needs to be improved, not that editing in companion (or radio) is necessary

Miami-Mike, 2021-07-29:
Then would you consider improving calibration by adding an option to specify that "I cannot center this particular analog input because it cannot be centered"? I replaced my LS slider with a push-button and I have to edit my calibration settings whenever I recalibrate in order to get an output range of -100% to +100%.

bigtwin, 2021-07-29:
This sound strange - have you replaced the tramsitters configuration and indicated that this channel is now a switch and not a slider anymore?

Why are you trying to blame the OTX team and refuse the responsibility for your own models? Its visible why 2.3.14 should be able at all to break a configuration which exists since 2.1.9. It rather seems like a simple misstake on your side. Beside this, I dont think Github is the place for posts like this.

Miami-Mike, 2021-07-29:
Okay bigtwin, right now in Companion 2.3.14 the options for LS are "Slider with detent" and "None". If they were to add the ability to specify that it's a switch and not a slider, as you suggest, then that would definitely work for me.

bigtwin, 2021-07-29:
I was not aware about this limitation. But this - on the other hand - would mean that OTX team needs to consider all kind of modifications to a given transmitter. At least I understand your problem now. 
I am curios how this ends up ...

Miami-Mike, 2021-07-29:
Notice that for a Taranis X9D+ the Hardware tab has provisions for S3 and for 6-positions switches, even for radios that don't come with those features. OpenTX has a long history of supporting hardware mods.

lshems, 2021-07-29:
Mike, you have to be grateful to the Devs for opentx. They do it all in their free time, as volunteers, no? So don't put up such a harsh tone. If someone is to blame, it's you, for using opentx in a way the Devs did not intend.

rotorman, 2021-07-30:
Out of curiosity I had a look at this (trying to get acquainted with Companion source code).

The change disabling editing calibration values stems from this commit: https://github.com/opentx/opentx/tree/1a8502e2840d42eec97853befa556fa47fbb25ba

Whereas the older revisions were using
https://github.com/opentx/opentx/blob/af0a10ec1284a323291c5d649d7f6eeb34739f02/companion/src/generaledit/calibration.cpp#L31
the newer version are using custom
https://github.com/opentx/opentx/blob/acbfef50a2191c17c49cfce64a9a875e07375219/companion/src/generaledit/calibration.cpp#L33

https://github.com/opentx/opentx/blob/acbfef50a2191c17c49cfce64a9a875e07375219/companion/src/helpers.h#L182-L204

for displaying the calibration values. There are no cell value readout functions implemented for TableLayout presently that would fetch the values back to https://github.com/opentx/opentx/blob/af0a10ec1284a323291c5d649d7f6eeb34739f02/companion/src/generaledit/generaledit.cpp#L273-L278
via
https://github.com/opentx/opentx/blob/af0a10ec1284a323291c5d649d7f6eeb34739f02/companion/src/generaledit/calibration.cpp#L84-L95
as done previously.

Miami-Mike, 2021-07-30:
I posted [a request to add the "Pot without detent" option to LS and RS](https://github.com/opentx/opentx/issues/8643). I think that would be an improvement no matter how this issue is resolved.