opentx/opentx #8640: Displaying Bluetooth device names instead of addresses

Gerd-St, 2021-07-28:
When searching e.g. for a head tracker or slave radio via Bluetooth discovery, all devices found are displayed just with their MAC address, what makes it difficult for selecting the right one as usually there are much more devices in range, e.g. phones. Please add an option for displaying the device names either in addition or as an option.

Oldgibber, 2021-08-28:
Big problem for old gibbers like me. Both my hearing aids pop up in Bluetooth discovery along with my mobile.

rotorman, 2021-08-29:
I did some sniffing of the FrSky Bluetooth module and radio microcontroller UART communication (for more see https://www.rcgroups.com/forums/showthread.php?3493859-RadioMaster-TX16S-16Ch-2-4ghz-Multi-Protocol-OpenTX-Radio-System/page662#post46878713 ), and as far as I could see, the FrSky Bluetooth module only relays the Bluetooth MAC address during discovery (`AT+DISC?` command), so there is not much an OpenTX radio can help/improve displaying the name of the Bluetooth devices unfortunately.

MlCHEL1234, 2021-08-29:
Can't this sloppy programming by FrSky be fixed by a "kernel patch" (similar to the CPU patches on the PC) in the OpenTx firmware?

kilrah, 2021-08-29:
OpenTX can't guess a name if none is given.