opentx/opentx #8642: Emergency mode triggered while modifying telemetry screens

geertvda, 2021-07-29:
On Radiomaster TX16S, 2.3.14:
In the attached model backup file, on model "Q12-X GRX", when working on the telemetry screens:
On the telemetry setup screen, after choosing "4+2 layout" and deselecting the sliders, the transmitter reverts to emergency mode when clicking on "setup widgets".
It happens every time, and it also crashes the simulator in exactly the same way.

[document1.zip](https://github.com/opentx/opentx/files/6903254/document1.zip)

GaryDT, 2021-08-18:
Yesterday I had the same thing happen on my Radiomaster TX16S running on 2.3.14, but it didn't happen every time, more like every other time. Also, on a few occasions when it didn't display emergency mode after deselected the sliders, I then restarted the radio to test it, but it had reverted back to the sliders being selected after rebooting, i.e., it didn't remember the change.

Also note that it didn't instantly display emergency mode, but actually did nothing visually for about 2 to 3 seconds... then displayed emergency mode.

Incidentally, my model plane wasn't powered up when this happened. When testing to see if this bug would show itself again later on yesterday evening, but with the model plugged in this time, it didn't happen for the few times that I tried, although I realise that could have just been a fluke that it didn't do it.

Thanks, Gary.

bitogre, 2021-08-21:
This looks like a duplicate of #8288 that I opened a while ago.  Nice to see someone else has reproduced this issue.

closed 2021-08-21 by 3djc