opentx/opentx #8643: Add the "Pot without detent" selection that S1 and S2 have to LS and RS sliders.

Miami-Mike, 2021-07-30:
I replaced the LS pot on my Taranis X9D+ with a pushbutton and can't calibrate it properly because there's no way to center it. Now, with recent Companion versions, I can't even work around that by editing the calibration settings.

But unless I'm mistaken, if the LS and RS pots had "Pot without detent" settings like S1 and S2 have then I could get the full -100% to +100% range without centering during calibration. So whether you restore the ability to edit calibration settings or not, I think it would be good for LS and RS to have "Pot without detent" options like S1 and S2 have.

3djc, 2021-07-30:
Makes sense

rotorman, 2021-07-30:
@Miami-Mike There might be a hardware solution to your issue. You could add two resistors to the analog channel with the switch in a voltage divider circuit fashion. The value of the resistors is not very relevant, it should not be too low for high current draw, but also not too high for better noise filtering, e.g. 2x 4K7 would be perfect.
One resistor from signal line to AGND, other from center pin to AVDD.
Then your switch pulling the signal line either to AGND or AVDD, does not matter. If you let the switch open, you have center position.
You could even add a 3-pos switch, one switch end would pull to AGND, other to AVDD.

Miami-Mike, 2021-07-31:
It's a pushbutton with two states. You're either pushing it or you're not.

[image]

You might be able to see in the picture that I also replaced S1 and S2 with 3-position switches. They're set up with resistor voltage dividers and after calibration the three positions return values of 100%, 0%, and +100%. S1 and S2 are set as "Pots with detent" on the Hardware page.

I discovered by experimentation with another OpenTX radio that, apparently, when S1 or S2 are set to "Pot without detent" then during the first step of calibration the center position is ignored. When S1 and S2 are set like that, it appears that only the end limits are considered and they return values linearly spaced from -100% to +100% with center having no significance.

That's what I need for LS. My experiments show that if S1 and/or S2 were replaced by pushbuttons then they could be made to return values of -100% and +100%, but LS and RS don't have that.

rotorman, 2021-07-31:
As you are customizing hw, why not just fork the source code and do the change in code as you like. You can change the LS and RS to have the functionality you like.
Building OpenTX from source is also not very daunting task, have a look at https://github.com/opentx/opentx/wiki/Compiling-OpenTx

3djc, 2021-07-31:
What type of x9d @Miami-Mike ?

Miami-Mike, 2021-07-31:
It's a FrSky Taranis X9D+, but I think this principle really should apply to OpenTX in general. When it's possible for the center position of a pot to lack significance, whether due to a modification or its assigned function, there should be an option to set it to "Pot without detent" so that its calibration is based solely upon its endpoints.

3djc, 2021-08-01:
Could you check this works ok ?
[firmware.zip](https://github.com/opentx/opentx/files/6912311/firmware.zip)

Miami-Mike, 2021-08-01:
I successfully installed the firmware file...

[image]

but I still only get "Slider" and "None" as choices for LS and RS...

[image]

[image]

and calibration still does the same thing. It behaves the way S1 and S2 do when set to "Pot with detent", not the way they behave when set to "Pot without detent".

The firmware you uploaded here appears to work the same as the 2.3.14 firmware I already have, aside from the different built options selected.

3djc, 2021-08-01:
I probably uploaded the wrong file then, let me check

3djc, 2021-08-01:
Here you go

[image]

[firmware.zip](https://github.com/opentx/opentx/files/6912832/firmware.zip)

Miami-Mike, 2021-08-01:
With your second version the "Slider w. det" option shows on my radio screen just like in your photo but there's no change in behavior.

If "Slider w. det" is meant to be the same thing as "Pot with detent" then that's not what I'm looking for, and in fact is no different from the way the slider already behaves. What I'm looking for is the equivalent of "Pot without detent", which on the Taranis X9D+ radio screen shows as simply "Pot".

[image]

3djc, 2021-08-01:
You obviously won't get Pot for a slider,, but "slider"entry should behave like pot does,

3djc, 2021-08-01:
No, can't be done wihtout chaning EEprom, we have only 1 bit for sliders; should have checked that first. Need to wait 2.4

Miami-Mike, 2021-08-01:
"Slider" currently behaves like "Pot with detent". If you change the label "Slider" to mean behavior like "Pot" (which is "Pot without detent" in Companion) then you'll change how some model setups behave. In some setups the "with detent" setting may be important because that setting ensures that the center position, as indicated by the detent, returns a value close to 0%. In other setups the center position is not important, such as when the potentiometer is used to control volume.

So it'll be a matter of choosing naming conventions and default settings that won't cause new problems. The name you chose, "Slider w. det." would work as a replacement name for what the current "Slider" setting does, but you wouldn't want "Slider" to be redefined as the setting where center is not significant (like the "Pot" setting).

Personally, I don't see the point in differentiating between the potentiometers on top that are controlled with knobs (S1 & S2) and the side potentiometers that are controlled with levers (LS & RS). They're all potentiometers, so why not provide the same setting options and associated labels for LS and RS that currently exist for S1 and S2?    

Anyway, thanks again for your effort!

3djc, 2021-08-01:
They are diffrents. Pots are CW, sliders toward top/center, but thats not the point, pots have 2 bits storage per pot (allowing 3 choi ces), while sliders only have one, and that cannot be changed without EEprom changes as I said, so 2.4

Miami-Mike, 2021-08-01:
I understand.