opentx/opentx #8646: R9M 2019 not binding in latest version update

martijnvwezel, 2021-07-31:
**Describe the bug**
Since that, I have update  OpenTX to the newest version (2021-07-23). The R9M 2019 module is not been able to bind with other devices. I have checked with multiple receivers (OTA and Slim), that are both set to ACCSS versions, like the radio module is set to the ACCSS version. The version that I used to have before this OpenTX version connected without problems.

**To Reproduce**
Steps to reproduce the behavior:
1. Making sure that both radio module and receiver have the ACCSS firmware and FLEX.
2. Put the receiver in bind mode (red and green LED are on)
3. go to radio settings -> external module -> select R9M -> 868 -> press bind
4. Nothing happens during the bind process

**Expected behavior**
The expected behavior is a binding between receiver and radio module

**General information**
 - OpenTX version: 2.3.14-otx (1e09791a)
 - Hardware: Opentx-tx16s
 - OpenTX Companion  2.3.13-otx on Ubuntu 18.04)

3djc, 2021-08-01:
Just tested here, working fine (well, once I flashed module and receiver to matching versions)

martijnvwezel, 2021-08-01:
@3djc  Can you elaborate on the exact versions you tested with? Or share the binaries/or location. So I can test your environment.

3djc, 2021-08-01:
As you said, R9M 2019, latest ACCST module (R9M-ACCST-20190117) and r9 firmware (R9-ACCST-20190201) on frsky website

martijnvwezel, 2021-08-01:
Why would it work before I have updated and now it does not? Some hardware failure in between? Is there a hardware test option? I have both updated with the same version you suggested using. (What I already was uploading)

3djc, 2021-08-01:
Not sure. Is you internal module off when you are using R9M ?

martijnvwezel, 2021-08-01:
Yes, I have also tested enabling the internal module but didn't make a difference.

brainbubblersbest, 2021-08-01:
Have you accidentially checked the Externalaccessmod Option?

martijnvwezel, 2021-08-04:
Yes, it was the ` Externalaccessmod Option` that was accidentally enabled. Thank you @brainbubblersbest you the best

closed 2021-08-05 by martijnvwezel