opentx/opentx #8653: Horus-like rados have communication problems with multiprotocol module.

prikrylm, 2021-08-05:
**Describe the bug**
I turned off both TX modules. If I want to start the external MPM, the LED flashes ("Slow flashing (on = 0.5 s, off = 0.5 s): a serial number has been selected, but no valid signal is visible on pin RX."), The radio displays "invalid protocol ".
If I start the external module with the internal module turned on, the MPM will start normally and work well, even if I turn off the internal module. If I switch the external module again (the internal module is switched off), the communication problem starts again. 

Problem start since OTX v2.3.13 (maybe v2.3.12?) Horus X12S with ISRM module as well as RadioMaster TX16S. V2.3.11 works well. There is no problem with otx 2.3.14 on Taranis (pre2019) radio.

**General information**
 - OpenTX version 2.3.13, 2.3.14
 - tested on RadioMaster TX16S and FrSky Horus X12S with ISRM

kilrah, 2021-08-05:
Could be that your module needs a bootloader update, there was something that was fixed in it for a similar issue quite a while ago.

prikrylm, 2021-08-06:
Bootloader in the module is from Flash-multi v0.6.1, so there is probably no newer one.

Skelly1001, 2021-08-17:
I have the HORUS X10s (not express model or upgraded transmitter module) and have the same problem.  I upgraded the boot loader to 0.6.1 and that didn't fix it.  In order to set up the MPM, I have to turn on the internal module, then turn on the external module, set the protocol I want to use and then turn off the internal module.  Until I do this steps, I'm left with the flashing red lights and a "invalid protocol" message.  After setting up the MPM for a plane, the previously established MPM setup works on subsequent sessions without going through this procedure.  Thanks to prikryllm for the post above.  I would haven't figured it out without his tip.  We definitely have a bug here.

3djc, 2021-08-21:
Issue confirmed, investigation in progress

PR #8680 (2021-11-10): Cleaner external module start
This fixes #8653
closed 2021-11-10 by 3djc